rstrouse/ESPSomfy-RTS #491: Sloping roller shutters inverted positions

thogitnet, 2024-10-31:
### Describe you new feature you'd like

First of all, thank you - mega project!

I have a sloping roller shutter which is controlled by the buttons on the controller. 

Means button up - the shutter goes up (extends) and covers the window. 
Pressing the down button lowers the shutter (retracts) and opens the window again.

This means that the status must be calculated in reverse.

At the beginning I thought that the inversion of the position was intended for this. But unfortunately it did not work. 

Enclosed you can see a warless picture from the WWW where such a sloping roller shutter setup combined with normal roller shutters is visible. [Link](https://www.bauen.com/bildergalerien/galerie/2,ausbau/9,fenster/72,Schraegrollladen/8/)

Perhaps there is an easy way to implement this. At the moment, the shutter status display is very confusing (open = closed and vice versa)

Thanks a lot.

rstrouse, 2024-10-31:
Try using invert the commands checkbox.

thogitnet, 2024-11-01:
I have also tested this variant. In this case, the display is correct. 
However, this means that the buttons are no longer correct.

On the remote control, pressing the top button extends the roller shutter upwards and closes the window. 
Pressing the down button does exactly the opposite.

Unusual at first - but totally logical as the button follows the direction of the shutter.

Comparable to roller shutters and awnings. - A retracted awning is closed - a retracted roller shutter is open.

I hope you can follow me and understand my problem ?

rstrouse, 2024-11-01:
I get it.  These change the definition of open and closed.  I'll have to look and see if there is a short direction to making it happen.

Out of curiosity is the motor mounted to the bottom or top?

thogitnet, 2024-11-01:
it is indeed mounted at the bottom and closes by pushing the lamellas upwards.

It is also exciting to see how the status is received by the Home assistant.