rstrouse/ESPSomfy-RTS #494: Added 2 new awnings lost 2 already connected screens

watgaan, 2024-11-09:
### Hardware

ESP32

### Firmware version

v2.4.6

### Application version

v2.4.7

### What happened? What did you expect to happen?

Got new awnings so I added these to the SomfyRTS using the procedure described in the instructions. This went fine program on remote jog program in SomfyRTS jog. Did both then I noticed that instead going from 6 to 8 devices in the shades tab I still had 6.
I lost 2 already configured devices, 2 screens.
When I then went to HA these were not added automatically I needed to reload HA. Then I got the new devices but the 2 screens were gone too.
When I checked the script I have for one of the screens that were removed I noticed the action to open screen was now populated by the newly added device....
As I am not sure what I do wrong I will for now not add the disappeared screens. Thanks for the support. Let me know when I can assist in providing more details. 

### How to reproduce it (step by step)

```bash
1. login to SomfyRTS on web interface
2. click on settings 'wheel'
3. go to Somfy
4. select tab shades
5. click on add shade
6. select awning by type
7. fill in unique number remote address
8. select room "Woonkamer"
9. provide name "Markies Keuken"
10. Leave all other settings default
11. click on add shade
12. click on pair shade
13. on telis remote press prog button
14. Awning jogs
15. Click on prog
16. awning jogs
17. click on link remote
18. press button on remote
19. remote not found as screen does not changes
20. added the second, same procedure also not able to pair remote
21. Noticed that I only had 6 items in the list on tab shades instead of 8. Missing screen Krijn and Screen left B&J
22. checked to see what max. number is, 32 according to the instructions
23. check HA to find new items not being added
24. reloaded HA
25. new awnings added
26. Screen Krijn grayed out
27. Screen left B&J gone
28. automation for screen left B&J now has as device Markies Keuken
29. opening this case
30. no logs found on the SomfyRTS device.
31. Thanks for the support!
...
```

rstrouse, 2024-11-12:
How did you get the v2.4.7 installed for the application without installing v2.4.7 for the firmware?  Also, please do not use any address from an existing remote.  This will not work.  If you use the address of an existing remote when adding a shade definition, eventually the remote and ESPSomfy RTS will stop working with that motor.  There are solutions out there that do that but ESPSomfy RTS does not operate that way.  It will coexist with the remote and you should be able to use either one.

watgaan, 2024-11-13:
Hi,

I installed version 2.4.7 via HA, it told me that there was an update and I installed. Did not see the update for the firmware, latest I saw was 2.4.6. My figuring was firmware and software can have different versions.
Your comment tells me this is not the case. I see when I check now the firmware there is a pre release of 2.4.7. 
I did not use the same addresses, used new numbers.
I have a back-up of the installed somfy motors before I tried to add the 2 new awnings. I can set that back, see how I can downgrade the HA integration to version 2.4.6 to match the firmware of the device. Then add the new awnings with a different unique ID again and hopeful that this then will work as it did before just an addition of Somfy motor next to the existing ones.
Let me know if you have a better way forward. Thanks for your support!
It is really appreciated 👍

rstrouse, 2024-11-14:
Ok the application version you are referring to is the plugin version for HA.  The firmware updates install I was referring to the application version that is displayed in the ESPSomfy RTS UI.  If you have further troubles upload a backup of the configuration.  You will find this in the system section under the firmware tab,

watgaan, 2024-11-14:
Aha I see, ESPSomfyRTS firmware 2.4.6 & Application 2.4.6. I will make backup live and then add both awnings and see what happens, I hope just normal behavior 2 new devices :) Will keep everyone posted on how it goes.

watgaan, 2024-11-17:
So I installed the back up back and had the 6 Somfy engines back, working as it should then restarted HA to get the devices back in HA. This worked only needed to delete 2 entities that related to the engines not available in back up.
Today had time to start again. Checked remote addresses. Picked a new number and added the shade. As soon as I had done this one of the existing shades was gone.
Did not pair the Somfy engine yet so that is the good news. So what is going wrong?

watgaan, 2024-11-18:
I looked at the backup files I have and noticed that the backup file made after adding the new awnings was using the same paired ID / number for the remote. I had not paired the them, yet. Also some other numbers were the same, figured out some are correct as they represent the type (awning = 3) and other and reference to which room etc.
With that in mind I made a third file where I added the 2 new awnings to the first backup figuring that by reading it back into ESPSomfyRTS it would add the new awnings. This did not work :(
I attached the backup files maybe they point out an issue? renamed the file type from .backup to .txt to be able to upload.
[ESPSomfyRTS 2024-08-20T15_03_24 - backup.txt](https://github.com/user-attachments/files/17796724/ESPSomfyRTS.2024-08-20T15_03_24.-.backup.txt)
[ESPSomfyRTS 2024-11-13T12_48_04 -backup.txt](https://github.com/user-attachments/files/17796726/ESPSomfyRTS.2024-11-13T12_48_04.-backup.txt)

rstrouse, 2024-12-01:
Restore this file then you should be able to add new shades.  Unfortunately, it looks like I have an indexer issue.   At the moment I am working on getting the upgrade to 3.x of the Arduino libraries and it has gotten a bit tough given the increase in library sizes.

[ESPSomfyRTS.2024-11-13T12_48_04.backup.txt](https://github.com/user-attachments/files/17970121/ESPSomfyRTS.2024-11-13T12_48_04.backup.txt)

watgaan, 2024-12-03:
Thanks will get this installed later today and keep you posted on the result. Thanks for the support.

watgaan, 2024-12-07:
Just tried to upload the provided file. Uploading went fine but same result, still only 6 somfy motors instead of 8. Got the 2 new Awnings but lost 2 existing screens (other ones then before). Restored to what I had and will wait with adding the new awnings when the indexer issue gets solved. Thank you very much for your support!

watgaan, 2025-01-05:
So after a while I had some time again to fiddle with Somfy :) 
I now see that in the file you provided only 6 motors are defined. That explains why it only shows 6 and I lose 2 screens again.
My aim is to have 8 motors in total, currently these are installed:
1,true ,  3,    248265,Markies Woonkamer
2,false,  0,    248281,Screen rechts
4,false,  0,    248288,Screen links
5,false,  0,    248269,Screen Krijn
7,false,  3,    248271,Markies Werkkamer
8,false,  0,    248272,Screen Aniek
Then I tried to add 2 awnings:
9,false,  3,    248292,Markies Woonkamer A
10,false,  3,    248298,Markies Keuken 
When I did this I lost 2 screens. I figured when I modify the backup file and add the 2 awings this would help me to get 8 devices. With the test file I created (
[ESPSomfyRTS 2024-11-17_test_met_2_markiezen.backup.txt](https://github.com/user-attachments/files/18310230/ESPSomfyRTS.2024-11-17_test_met_2_markiezen.backup.txt) )
I hoped to get it solved, unfortunately this did not solve it. Any thoughts on what I do wrong in the backup file?
Thanks for the support.