rstrouse/ESPSomfy-RTS #495: Is it possible to add sun sensors, too

ErikSteiner, 2024-11-12:
### Describe you new feature you'd like

Hey there,

I have three **Somfy Sunis WireFree II io**, that are paired with my Somfy Tahoma Box. Is it possible to pair these sensors with your solution, or do you know any open source project, that makes it possible to do so?

Thank you very much.

rstrouse, 2024-11-12:
Yes.  These operate the same way as a wind sensor.  Do not forget to set the Has Sun Sensor checkbox as well.

https://github.com/rstrouse/ESPSomfy-RTS/wiki/Configuring-the-Software#linking-sensors

ErikSteiner, 2024-11-13:
Wow, that's awesome. I'm so excited to start the project. I've reached the maximum number of scenarios with my Tahoma box and haven't implemented everything yet. Your project will remove this barrier for me.

Nickduino, 2024-12-04:
> Somfy Sunis WireFree II **io**

@rstrouse They seem to use io-homecontrol, though

ErikSteiner, 2024-12-04:
Is this a good, or a bad thing?

Furthermore, I have two window sensors from Somfy. That can detect an open or closed window (Somfy Slim Opening Detector io 1811638). So I would like to extent my question to this sensor.

Nickduino, 2024-12-04:
Somfy uses two protocols:

- RTS, its historical protocol from 1997;
- IO Homecontrol, a more recent development used by Somfy, Velux, Atlantic, Sauter and maybe others.

IO is objectively a better protocol (it's more secure and two-way, meaning the equipment can report their state) but it's extremely closed while the RTS protocol has been reverse-engineered years and years ago. Like its name suggest, ESPSomfy-RTS works only with RTS devices.

The easiest solution for you would probably to connect your Tahoma to Home Assistant (or another home automation solution) and create the scenarios in it.

rstrouse, 2024-12-06:
Actually, if you have RTS motors the IO version of that sensor will not work with them.  The way Sunis works (yes there is an RTS version of it) is that is broadcasts a sunny or shady condition to all motors that are paired with it.  If it is not paired (and if IO it cannot be paired with an RTS motor) with the motor it will not do anything.

Unfortunately, IO Homecontrol hubs used to require a connection to the interweb to operate.  I don't know if this is true anymore though.  Why anyone would want the hardware in their home to call out to an external server as part of the control chain is a flat mystery to me.

Nickduino, 2024-12-06:
> Unfortunately, IO Homecontrol hubs used to require a connection to the interweb to operate. I don't know if this is true anymore though. Why anyone would want the hardware in their home to call out to an external server as part of the control chain is a flat mystery to me.

HA can control a Tahoma Switch locally for some tasks, I don't remember which.