rstrouse/ESPSomfy-RTS #497: Suggested delay between commands

Vitek-1977, 2024-11-17:
### Describe you new feature you'd like

Hello, first of all let me thank you very much for the project, it is really great. Btw is there any way to support your effort like Buy Me Coffee? I would like to contribute that way.
Now to the question - went for controlling my Somfy shades through Apple Home, everything works excellent except controlling two or more shades at a time - in one room I have two separate windows and the command - close the shades - worked for one only, obviously ESP does not handle both requests at one time. So far I put 2 secs delay for the second shade, but now I have entire home in the project and two seconds for every command - and I am sending two for each shade (open/close and tilt) would be too much so if you went through this, what would be suggested delay between the mqtt commands to be sent to control the shades?

Once again many thanks

Vitek

rstrouse, 2024-11-17:
Actually your problem is probably not what you think.  The problem is most likely tied to the radio reset on your motors,  Every RTS motor will decode every frame then decide whether or not to act on it.  However, during this period some motors stop receiving so while it is processing the last frame it heard it will not hear the frame that is actually for it.  This silence can last a couple of frames long and not recover in time to hear the radio preamble it should have heard in the first place.

To fix this simply add a couple of command repeats in the ESP Somfy configuration for the motors.  Increase the number of repeats until it hears every command.  The ESP32 is capable of sending 10 frames per second.

Vitek-1977, 2024-11-18:
Ok, will go this way, Thank you again!

Vitek-1977, 2024-11-25:
Woked very good, thanks for the help!

Vitek-1977, 2024-11-25:
Solved

closed 2024-11-25 by Vitek-1977