rstrouse/ESPSomfy-RTS #498: somfy dexxo optimo + KEYTIS RTS remote

KaizenKuba, 2024-11-18:
### Describe you new feature you'd like

Hello,
I have 2 garage gates with somfy dexxo optimo + KEYTIS RTS remote and Entry gate (dont know motor) but also works on KEYTIS RTS remote.
My logs do not show nothing. I changed to 80 bit and also no results.
I found in other issue workflow with "virtual button" to check but not sure how to do it. should I pair it like in this tutorial https://www.youtube.com/watch?v=_EGp2yuD4Js but intead of physical button use virtual button in ESPSomfy?

what is correct way to add my 1-button gates to ESPSomfy-RTS?

PS. Huge respect for this project. I hope you will help me to connect it and you will bring the 21st century to my home.

rstrouse, 2024-11-18:
Please verify your wiring and configuration for the transceiver.  If it is correct and the transceiver is set up properly it should be hearing the Keytis remote.   There are others here that are using the one-button remotes.

avantgarde0707, 2024-12-29:
I have same issue. All other Somfy shutters works fine only dexxo optimo + KEYTIS RTS remote not! @KaizenKuba: Any solution  so far?