rstrouse/ESPSomfy-RTS #499: [Questions] Measuring Motor Timings and Migration

wilriker, 2024-11-20:
### Describe you new feature you'd like

Hi @rstrouse!

I am in the process of setting up ESPSomfy-RTS. Actually I am converting over from an old and unreliable `pushstack`-based implementation with Raspberry Pi and SI4010 module. And I have two questions:

## Motor Timings
I was measuring my motor timings and as a true nerd does that I would not rely on simply using a stop-watch. I recorded the audio of each motor while it was closing and opening and afterwards measuring the time in software.

As of right now I entered the pure motor-on-time for these movements. But I was wondering if I need to add the ~700ms delay between pressing the button on an existing remote and the motor actually starting to turn? The more I think about the more I get confused if that makes sense or not. So can you please share your opinion?

## Migration
As mentioned I plan to move over from `ser4010_somfy` (as created by `pushstack` - or `dimhoff` here on GitHub) and thus I have already a full set of "virtual" remotes. I want to spare me the extra effort of un-pairing my SI4010 based solution (as I mentioned this became unstable over the years - mainly due to Kernel changes) and just reuse those for ESPSomfy-RTS.
My idea is:
Copy over the Remote Address and Rolling Code from the existing setup for each roller shade and set it as paired.

There is the remaining difference that for ESPSomfy-RTS the `Key` is hard-coded to `0xA7` while the existing ones get changed with every invocation of `ser4010_somfy` to have the 4 LSB increase with the Rolling Code. AFAICT this difference should not pose a problem, right?

Cheers
Manuel

rstrouse, 2024-11-21:
The key should not make a difference.  While it does change between fames it is not used for frame decoding.  In ESP Somfy the key does change but it really does not matter so long as the checksum is calculated properly so there is no need to worry about that.

wilriker, 2024-12-02:
I did not get it to work with the existing address but I might have made a mistake translating them from Hex -> Dec (I have no idea in which endianess they had been written out in the corresponding file). So I eventually registered new remotes and I'm really happy on how great it works. So thanks for this project!