rstrouse/ESPSomfy-RTS #500: ESP32-C6 WIFI AP not appearing

nicholasshay, 2024-11-23:
### Hardware

ESP32

### Firmware version

v2.4.6

### Application version

v2.4.6

### What happened? What did you expect to happen?

Hi. Does this work with an ESP32-C6 or do I need a different chipset? I have 'successfully' flashed multiple times using Esptool and I can't get the ESPSomfy RTS WIFI AP to appear.  cheers

### How to reproduce it (step by step)

```bash
I am using bash on MacOs. 

1. connect esp32-c6 via usb c whilst holding down boot button
2. execute bash-3.2$ esptool.py --port /dev/tty.usbmodem1401 write_flash 0x0 /Users/nickshay/Downloads/SomfyController.onboard.esp32.bin
3. push reset
4. go the wireless settings on MAC and search for ESPSomfy RTS SSID. Also tried entering manually with no security. It could not find the network.
```

### Logs

```bash
bash-3.2$ esptool.py --port /dev/tty.usbmodem1401 write_flash 0x0 /Users/nickshay/Downloads/SomfyController.onboard.esp32.bin
esptool.py v4.7.0
Serial port /dev/tty.usbmodem1401
Connecting...
Detecting chip type... ESP32-C6
Chip is ESP32-C6 (QFN40) (revision v0.0)
Features: WiFi 6, BT 5, IEEE802.15.4
Crystal is 40MHz
MAC: 40:4c:ca:ff:fe:57:7d:cc
BASE MAC: 40:4c:ca:57:7d:cc
MAC_EXT: ff:fe
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x003effff...
Compressed 4128768 bytes to 928823...
Wrote 4128768 bytes (928823 compressed) at 0x00000000 in 9.2 seconds (effective 3603.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

rstrouse, 2024-11-23:
The ESP32-C6 series chips are not yet supported.

stereotone, 2024-11-24:
I have the same issue with the ESP32 WROOM. No AP appears....

rstrouse, 2024-11-24:
@stereotone your issue is very different than @nicholasshay's please download `SomfyController.onboard.esp32.bin.zip` archive and extract its contents.  You will find a file named `SomfyController.onboard.esp32.bin` file inside this archive.  You must flash with this file to onboard your board.

Please review the documentation below.
https://github.com/rstrouse/ESPSomfy-RTS/wiki/Installing-the-Firmware

zipokotom, 2024-11-26:
Hi, I have the same issue as stereotone, and can’t find the onboard/archive files you speak of.

nicholasshay, 2024-11-26:
> The ESP32-C6 series chips are not yet supported.

Silly me! OK, I am now trying with a ESP32S3. Firmware install went fine but I am stuck in a boot loop. Says SHA-256 comparison failed?

nicholasshay, 2024-11-26:
> > The ESP32-C6 series chips are not yet supported.
> 
> Silly me! OK, I am now trying with a ESP32S3. Firmware install went fine but I am stuck in a boot loop. Says SHA-256 comparison failed?

ignore that. I am up and running! cheers

rstrouse, 2024-11-27:
> Ok. I can see Up and My + Up commands from my somfy situo RTS 5 II remote appearing in the log. I cannot get it to recognise the 'down' button though. Any clues as to what I should check? ..Nick

That is odd.  Can you upload a log of pressing the up, then my, then down.   The byte mapping for the Situo RTS remotes are not special in any way and is currently being used by many.  Do you have the variant with the thumbwheel?

nicholasshay, 2024-11-30:
I don't have the thumb wheel, no. I have tried using a different PIN for Rx but then it stopped working all together. 

I am using this board. https://thepihut.com/products/esp32-s3-microcontroller-development-board

cheers 

[image]

rstrouse, 2024-11-30:
The my + up and the sun flag do not appear to be from your remote.  Do you have neighbors nearby where these may be coming from?  If I were to guess, the address ending in 838 may be from your remote.

nicholasshay, 2024-12-01:
Well that's interesting. Both my neighbour and I have markilux awnings with somfy remotes, but not convinced it is that as we are into winter and they don't get used in the winter. Perhaps another neighbour, very weird. As an aside I just tested my markilux awning remote and ESPSomfy picks up the up, down and why presses perfectly (address ending 712). But ESPSomfy only recognises my shade remote (the address ending 838) Up command. Very weird. 

[image]

nicholasshay, 2024-12-01:
[image]

rstrouse, 2024-12-01:
That remote is probably the most common remote used so lets do some things and see if we can troubleshoot this.

First, how are the batteries in the remote?  The signal will do weird things when the battery level gets low.  For instance, it is not uncommon that it will properly send the first frame properly then not send good frames until it has not sent any for a while.

Next open the transceiver settings and set the RX bandwidth to 50 then increase the RX Bandwidth by 100 until it sees every button.  Save the radio settings after each change you make.  It may be that this remote is just a little bit out of tune.  You do not need to pop back and forth to the log tab as you should see the RSSI increment each time it hears a frame from the button.
[image]

nicholasshay, 2024-12-02:
Now am struggling to pair the shade. Cannot get the Tx to work when I press prog. I don't have a pin layout for my cc1101. Do all the 8 pin CC1101s use the same layout? Any other ideas?

rstrouse, 2024-12-02:
No they do not all have the same pin layout.  However, if you take a picture of it I may be able to identify it.  If it looks like this one with the screen printing then the layout is what is documented in the wiki.  If it is follow the instructions after the picture.
[image]

Find pin 3 on the transceiver it will be the pin to the immediate right of the pin labeled pin 1 (on the bottom row) and note which pin it is connected to on the ESP32.   Then verify this is the pin you have selected as the TX pin in the radio configuration.

nicholasshay, 2024-12-02:
My transceiver is exactly the same and I have validated the Tx pin configuration. 

[image]

rstrouse, 2024-12-02:
First make sure you have not set your TX Power higher than 10 for that transceiver.   Next unscrew the antenna and make sure it has a pin in the middle of it.  I received one of these early on that shipped with an RP-SMA instead of an SMA antenna.  Which pin are you using for TX and which ESP32 board did you finally land on?

nicholasshay, 2024-12-02:
The antenna looks ok. TX Power is set to 10 Db (I have tried 7). I have connected PIN 3 on the CC1101 to PIN 38 on the ESP32S3. I have tried multiple pins actually. I landed on this one. https://thepihut.com/products/esp32-s3-microcontroller-development-board

rstrouse, 2024-12-03:
This may not have anything to do with the pin selections.   Some motors will respond with a jog to indicate they are in prog mode but not actually store an address if the address list in the motor is full.  You may need to factory reset your motor. 

Below are the pins I used on the S3 Mini but it is the same chip series.
https://github.com/rstrouse/ESPSomfy-RTS/wiki/Mini-ESP32s3-Device-Build#solder-e07-m1101d

nicholasshay, 2024-12-22:
I have multiple somfy motors for my blinds (battery powered) and a powered motor for my awnings. None of them work. I am struggling to believe that I need to reset all the motors. There must be another problem?

optimoto, 2024-12-27:
Hi, I am having the same boot loop problem that @nicholasshay had previously. I can't figure out how to make this work. Do I need to disable secure boot loader somehow?

optimoto, 2024-12-27:
Nevermind...working now. The thing that worked for me was installing ESPhome on the device, connecting to it through wifi, then doing an OTA update from the device's web interface. For some reason, installing the firmware using the web.esphome.io kept giving me SHA 256 mismatch errors and endless boot loops.

Tsury, 2025-12-09:
If anyone's still interested in flashing this on a C6, I made a [fork specifically for it](https://github.com/Tsury/ESPSomfy-RTS).

shailensobhee, 2026-03-30:
For all those who also have the ESP32-C6 board, mine is ` esp32-c6-devkitc-1-n-4` with `ESPP32-C6-WROOM-1` written on it, I found this repo to work: https://github.com/cjkas/ESPSomfy-RTS
I also made some improvements to that code above to work out-of-box on the C6 board I have.