rstrouse/ESPSomfy-RTS #501: Support for tilting vertical blinds (motor on the right)

hajar97, 2024-11-29:
### Describe you new feature you'd like

I have blinds with vertically hanging tiles, motor located on the right, blinds opening all the way to the right, and closing all the way to the left. They are also have integrated tilting mechanism. 

Since tilt is only supported for blinds, I have chosen this type of motor. However, all the HA icons used make it look like as if it is a horizontal blind, while in fact it is vertical. 

Would it be possible to extend the functionality of ESPSomfy RTS to also support vertical blinds with all corresponding HA icons and UI to match?

rstrouse, 2024-12-01:
You can override the motor type in HA.  I was hoping to implement the latest v3 of the Arduino Core but it appears that the core libraries have expanded so much that they do not fit on the ESP32 flash partition anymore.  Then add some additional tilt functions for vertical blinds as well as some directional components for those where the cover feeds from the bottom.

I am slowly adding libraries until I find the one with the size problem.

hajar97, 2024-12-01:
Thank you. Not sure how to override the motor type in HA. Anyway, its not a big deal. It works flawlessly. I am happy to wait until you can add support for vertical blinds later on.

Peethiplatsch, 2025-05-04:
I also have a question regarding this. You wrote thst I can override the type in HA, fair enough. What I am curious about is the tilting. As far as I understand is the engine an open/close button control. Means tilting starts with that as well. Can you set this up properly with a % indicator and slider as well?

Peethiplatsch, 2025-07-15:
@hajar97 i have a question. Is the tilting working properly with your motor? Mine is not working with this setup. When opening the slats its always also opening the blinds completely. This is really annoying since I can never set them to 0%. This is because esp somfy is not sending a stop signal. Do you have the same issue? If so, have you fixed it?

hajar97, 2025-07-15:
> Peethiplatsch

Hey I am not home right now and cannot look deeply into it. But I can confirm that I definitely do not have that issue. Blinds tilt perfectly and stop correctly. The only issue I have is that same % of tilt results in slightly different actual tilt depending on what was the starting position of my curtain. But the difference is small enough to just ignore it most of the time. So overall I am super happy with how it well it works. It basically never fails and always does its job well. And I do use it in daily automations, so it is running many times a day.