rstrouse/ESPSomfy-RTS #503: Somfy Situo 1 io Pure II Not detected

Hollako, 2024-12-01:
### Hardware

ESP32, ESP32-S3

### Firmware version

v2.4.7

### Application version

v2.4.7

### What happened? What did you expect to happen?

Hello  Gents,

First of all thank for this great project and for your efforts. I made a device and downloaded the ESPSomfyRTS on it and everything works good with many remotes of curtains and shades and all worked well as expected, except one type of Remotes which is the Situo 1 io Pure II, the ESPSomfyRTS is not detecting the frequency of these remotes at all and also when i enter the program mode to any of the shutter and then press the Prog button in ESPSomfyRTS it's not responding (jogging) at all. tried to changed the protocal and frequencies without success. Just i need to know what might be the problem or it's an old remote since the shutter are installed long time ago while the curtains and shades are recently installed.

Thank you in advance

Regards

### How to reproduce it (step by step)

```bash
1. Go to...
2. Click on...
...
```

rstrouse, 2024-12-01:
Unfortunately, the Situo 1 io is not an RTS protocol remote.  This solution does not support the IO Home control protocol.  Sorry.

Hollako, 2024-12-01:
Hello, Thanks for quick reply and well noted. it means another frequency or protocol

Thanks again.