rstrouse/ESPSomfy-RTS #504: ESP32-C3 watchdog kicks in

wilriker, 2024-12-02:
### Hardware

ESP32-C3

### Firmware version

v.2.4.7

### Application version

v2.4.7

### What happened? What did you expect to happen?

When I try to enable the radio it states `Radio Pins Configured!` on the console. A few seconds later the watchdog kicks in rebooting it.

### How to reproduce it (step by step)

```bash
1. Go to Radio Settings
2. Select the appropriate pins according to this schematic: https://www.imiconsystem.com/wp-content/uploads/2023/12/esp32-c3-super-mini-01.webp (I chose GPIO 3 for TX and GPIO 2 for RX - no other combination seemed to work at all)
3. Click Enable Radio and Save Radio
4. Observe the watchdog reset on serial console

This also happens on v2.4.6 but there is no output on serial console.
```

### Logs

```bash
Save Radio Settings SCK:4 MISO:5 MOSI:6 CSN:7 RX:2 TX:3
Applying radio settings Setting Data Pins RX:2 TX:3
Setting SPI Pins SCK:4 MISO:5 MOSI:6 CSN:7
Radio Pins Configured!
E (140224) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (140224) task_wdt:  - loopTask (CPU 0)
E (140224) task_wdt: Tasks currently running:
E (140224) task_wdt: CPU 0: loopTask
E (140224) task_wdt: Aborting.

abort() was called at PC 0x42045ee3 on core 0
Core  0 register dump:
MEPC    : 0x403827b8  RA      : 0x40387006  SP      : 0x3fc9e370  GP      : 0x3fc8e400
TP      : 0x3fc7ea30  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x3fc9e39c  S1      : 0x3fc9e39c  A0      : 0x3fc9e3a8  A1      : 0x3fc9e38a
A2      : 0x00000000  A3      : 0x3fc9e3d5  A4      : 0x00000001  A5      : 0x3fca2000
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x3fca2630  S3      : 0x7fffffff
S4      : 0x3fc8fae4  S5      : 0x3fc8fad4  S6      : 0x3fc8fadc  S7      : 0x3fc8fad4
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00001801  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000
MHARTID : 0x00000000

Stack memory:
3fc9e370: 0x00000000 0x00000000 0x3fc9e388 0x4038c624 0x3fc8fadc 0x3fc8fad4 0x3fc80030 0x3fc90d20
3fc9e390: 0x3fc9e39c 0x3fc90d3c 0x3fc9e388 0x34303234 0x33656535 0x00000000 0x726f6261 0x20292874
3fc9e3b0: 0x20736177 0x6c6c6163 0x61206465 0x43502074 0x34783020 0x35343032 0x20336565 0x63206e6f
3fc9e3d0: 0x2065726f 0x00000030 0x3c0f0000 0x3d859198 0x600c2000 0x3c0fa000 0x3fca2000 0x42045ee6
3fc9e3f0: 0x00000008 0x00000001 0x00000019 0x000223c0 0x00000004 0x00000003 0x3fcaaffc 0x00000001
3fc9e410: 0x3c0f4000 0x3c0f4000 0x3fc95000 0x00000001 0x00001881 0x80000007 0x3fca256a 0x403801ee
3fc9e430: 0x00000001 0x3fc9e438 0xffffffff 0x3fca76f4 0x3fca76f4 0x00000001 0x3fc9e44c 0xffffffff
3fc9e450: 0x3fcaa000 0x3fcaa000 0x00000000 0x3fc9e460 0xffffffff 0x3fc9e460 0x3fc9e460 0x00000000
3fc9e470: 0x3fc9e474 0xffffffff 0x3fc9e474 0x3fc9e474 0x00000000 0x3fc9e488 0xffffffff 0x3fc9e488
3fc9e490: 0x3fc9e488 0x00000000 0x3fc9e49c 0xffffffff 0x3fc9e49c 0x3fc9e49c 0x00000000 0x3fc9e4b0
3fc9e4b0: 0xffffffff 0x3fc9e4b0 0x3fc9e4b0 0x00000000 0x3fc9e4c4 0xffffffff 0x3fc9e4c4 0x3fc9e4c4
3fc9e4d0: 0x00000000 0x3fc9e4d8 0xffffffff 0x3fc9e4d8 0x3fc9e4d8 0x00000000 0x3fc9e4ec 0xffffffff
3fc9e4f0: 0x3fc9e4ec 0x3fc9e4ec 0x00000000 0x3fc9e500 0xffffffff 0x3fc9e500 0x3fc9e500 0x00000000
3fc9e510: 0x3fc9e514 0xffffffff 0x3fc9e514 0x3fc9e514 0x00000000 0x3fc9e528 0xffffffff 0x3fc9e528
3fc9e530: 0x3fc9e528 0x00000000 0x3fc9e53c 0xffffffff 0x3fc9e53c 0x3fc9e53c 0x00000000 0x3fc9e550
3fc9e550: 0xffffffff 0x3fc9e550 0x3fc9e550 0x00000000 0x3fc9e564 0xffffffff 0x3fc9e564 0x3fc9e564
3fc9e570: 0x00000000 0x3fc9e578 0xffffffff 0x3fc9e578 0x3fc9e578 0x00000000 0x3fc9e58c 0xffffffff
3fc9e590: 0x3fc9e58c 0x3fc9e58c 0x00000000 0x3fc9e5a0 0xffffffff 0x3fc9e5a0 0x3fc9e5a0 0x00000000
3fc9e5b0: 0x3fc9e5b4 0xffffffff 0x3fc9e5b4 0x3fc9e5b4 0x00000000 0x3fc9e5c8 0xffffffff 0x3fc9e5c8
3fc9e5d0: 0x3fc9e5c8 0x00000000 0x3fc9e5dc 0xffffffff 0x3fc9e5dc 0x3fc9e5dc 0x00000000 0x3fc9e5f0
3fc9e5f0: 0xffffffff 0x3fc9e5f0 0x3fc9e5f0 0x00000000 0x3fc9e604 0xffffffff 0x3fc9e604 0x3fc9e604
3fc9e610: 0x00000000 0x3fc9e618 0xffffffff 0x3fc9e618 0x3fc9e618 0x00000001 0x3fc9e62c 0xffffffff
3fc9e630: 0x3fcabda8 0x3fcabda8 0x00000000 0x3fc9e640 0xffffffff 0x3fc9e640 0x3fc9e640 0x00000000
3fc9e650: 0x3fc9e654 0xffffffff 0x3fc9e654 0x3fc9e654 0x00000005 0x3fc9e668 0xffffffff 0x3fcaee8c
3fc9e670: 0x3fcb304c 0x00000000 0x3fc9e67c 0xffffffff 0x3fc9e67c 0x3fc9e67c 0x00000000 0x3fc9e688
3fc9e690: 0x00000000 0x00000000 0x00000000 0x3fc9e6a0 0xffffffff 0x3fc9e6a0 0x3fc9e6a0 0x00000000
3fc9e6b0: 0x3fc9e6b4 0xffffffff 0x3fc9e6b4 0x3fc9e6b4 0x00000001 0x00000001 0x00000000 0x0001ffff
3fc9e6d0: 0x00000000 0xb33fffff 0x00000000 0x00000000 0x3fc9e6dc 0x00000000 0x00000000 0x00000000
3fc9e6f0: 0x3fc9e6f4 0xffffffff 0x3fc9e6f4 0x3fc9e6f4 0x00000000 0x3fc9e708 0xffffffff 0x3fc9e708
3fc9e710: 0x3fc9e708 0x00000001 0x00000001 0x00000000 0x0001ffff 0x00000000 0xb33fffff 0x00000000
3fc9e730: 0x00000009 0x3fca3964 0x3fca39cc 0x3fca3a34 0x00000000 0x00000000 0x00000001 0x00000000
3fc9e750: 0x00000000 0x00000000 0x420d5ede 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000

ELF file SHA256: 677ce0d72002f1dc
```

rstrouse, 2024-12-02:
It looks like you may have used some pin numbers from the C3 that are labeled differently.   A lot of these boards put a number next to the pin which is simply a digital pin number and not the corresponding GPIO number.  For instance see below D3 is not GPIO3.  Which ESP32c3 do you have?

[image]

wilriker, 2024-12-02:
I do have a Tenstar Robot ESP32-C3 Super Mini.

wilriker, 2024-12-02:
Quickly tried with the pinout in your image above but the result is still the same.

rstrouse, 2024-12-02:
There are reports of these boards out there that have either no flash or flash issues.  SPI0/1 is typically reserved for external flash.

However, I would remap your pins to the following.

SCK:GPIO8 MISO:GPIO9 MOSI:GPIO10 CSN:GPIO6 
RX:GPIO4 TX:GPIO3

wilriker, 2024-12-03:
I tried your suggstion but the result keeps the same. Watchdog gets triggered.

I ordered a second ESP32-C3 yesterday but that only arrives mid next week.

wilriker, 2024-12-06:
Today I received a replacement ESP32-C3 and that one works. But I have to use my initial SPI pin configuration. With the one you suggested none of my C3's even boot (as soon as anything is connected to GPIO9).

Thanks, I can now continue with my minification build. :-) And sorry for the false alarm.

closed 2024-12-06 by wilriker

Gizmo-HA, 2024-12-07:
Hello,

I have the ESP32-C3 than you.

wilriker, 2024-12-07:
Hi @Gizmo-HA 

Pinouti eventually used is the one for SPI from the image I linked to in my original post. Rx and Tx are on GPIO 02 and 03 respectively. 

So effectively
SCK:4 MISO:5 MOSI:6 CSN:7 RX:2 TX:3

Where the connections have to go on your CC1101 you'd have to look up in the wiki since I have an E07-M1101D-SMA.

I don't know why it did not work initially - after retrying for the about 10th time it now also worked on the initial C3.

Cheers 
Manuel