rstrouse/ESPSomfy-RTS #505: Inverted percentage (yes I know your opinion and I'm with you)

Kev0123, 2024-12-05:
### Hardware

ESP32

### Firmware version

v2.4.6

### Application version

v2.4.7

### What happened? What did you expect to happen?

First off all, I'm with you, a percentage of open is stupid.

Because of that I played around with another integration (Adaptive Covers) and stumbled over this nonsens that 100% is a fully recracted awning.

Because of this I played around with the invert position to get home assistant an the webGUI a matching value. (of open % 🤮)

Somehow after a reboot of home assistant the state is now inverted and the invert position checkbox won't change anything in home assistant.

I do not know what caused this but I can't change the % behavior in homeassistant anymore with the invert checkbox in GUI, which was previous possible.

Am I lost? Why won't home assistant retract the awning if I set the value to 100% and invert the percentage in ESPSomfy GUI? Will this even be the case? As before the restart of HA it was the case. I was able to invert the behaviour of HA percentage with the ESPSomfy setting.

[image]
[image]

Now the slider in HA will just use the percentage no matter the setting in GUI.

invert off: HA 100% open will extend the awning 100%
Invert on: HA 100% open will extend the awning 100%

How to solve this that HA will act like before and usr its "great" open-percentage?

Regards

### How to reproduce it (step by step)

```bash
E.g.: 
inverse position off. Awning at 23%. HA slider 23% open

Inverse position on. Awning at 23%. Put HA slider to 77% nothing happens. (So far so good)-> put slider in HA to e.g. 80% it will switch state to 20% open and retract 3 %
```

rstrouse, 2024-12-06:
Simply go into home assistant and reload the integration.  It does not requery the configuration until it needs to reload from a reboot or forcibly reloading the integration.

Kev0123, 2024-12-06:
This does not work, thats the reason of this post :-)

Kev0123, 2024-12-17:
@rstrouse no other suggestion on this?

Kev0123, 2025-03-27:
@rstrouse could you have a look at this?