rstrouse/ESPSomfy-RTS #508: My Button Not Working in WebUI

Snelk, 2024-12-15:
**Hardware:** 
ESP32C3
E07-M1101D
Telis 4 RTS (https://www.remotecontrol-express.com/remote+SOMFY+:+TELIS-4-RTS-OLD)
System II Roll-Up (LT-30)
**Firmware version:** v2.4.7
**Application version:** v.2.4.7

**Issue Description:**
I was able to pair my shades without problems, but when I press the My button within the Web UI, nothing happens. The up and down keys function as expected and if I go to the virtual remote page, the "toggle" button there functions as I would expect the middle/my button to on my physical remote.

Behavior within home assistant is also odd. Up/down/stop work, tilt close Opens the slats and tilt open raises the shade up.

Any ideas what might be wrong? 

**Expected Behavior:**

- Shade Up: Up: Nothing, My: Lower to 100% Down and tilt/slats remain closed, Down: Lower to 100% Down and tilt/slats open
- Shade in Motion: Any button stops the shade
- Shade Partially Down: Same as shade up, but Up button will raise the shade fully.
- Shade 100% Down: Up: Raise all the up, My: Closes tilt/slats if open, Down: Opens tilt/slats if closed
**Shade Config:**
[image]

My shades and remote were installed something like 20 years ago. The physical remote only sends 3 commands; Up, Down, and My:
[image]

Any help would be appreciated!

rstrouse, 2024-12-15:
That my button on your remote serves a dual purpose.  If the motor is moving it should stop the movement when the command is sent.  If it is stopped then it should move the my position but only if it has been set in ESPSomfy RTS.   Please set a favorite using the functionality described below.  If your motor does not support the setting of a favorite, then you can use the Simulate Favorite Position checkbox but it must be set in ESPSomfy RTS.

In the 56-bit protocol, the My button and Toggle button send the same command.

Also, integrated tilt mechanisms are a bit tricky as 100% is actually 50% of the total movement.   An open slat is one that is in between the open and closed positions in HA.  The cover component is a bit goofy like that.  

Calling Open Tilt without setting a slat position causes HA to send a single command since the motor actually knows nothing about your slats.  It only knows to roll up or down and will not take into account the stop command that must be issued when the slats get to their open or closed position.

https://github.com/rstrouse/ESPSomfy-RTS?tab=readme-ov-file#setting-a-favorite

Snelk, 2024-12-15:
Thank you! 

The My button works now.

If the My and Toggle command is the same using the 56-bit protocol, any idea why the toggle button worked but the my button did not before setting the position within ESPSomfy? 

Yeah, Home Assistant can definitely be weird at time. I was able to get that working as well now that the My button works. The "Close Tilt" option in HA will Open the slats and the Stop button will close the them. Kind of jank, but it does what I need it to!

Anyway, thank you again for the help and all the work you've done on this project! Being able to control all my blinds without having to stand in the one, und ONLY ONE, spot that the Telis remote can reach them all is wonderful!

rstrouse, 2024-12-16:
> If the My and Toggle command is the same using the 56-bit protocol, any idea why the toggle button worked but the my button did not before setting the position within ESPSomfy?

This is because the my button when processed by ESPSomfy RTS is a bit smarter than the remote.   It first checks to see if the motor is in motion and if the my position needs to triggered.  Toggle is a bit of a different command that is typically reserved for 1 button remotes for garage doors, gates, and dry contracts.

Snelk, 2024-12-17:
Gotcha, makes sense.

Thank you again for the help!

closed 2024-12-17 by Snelk