rstrouse/ESPSomfy-RTS #509: Fast successive commands lead to skipped TX

wilriker, 2024-12-16:
### Hardware

ESP32-C3

### Firmware version

v2.4.7

### Application version

v2.4.7

### What happened? What did you expect to happen?

I have my setup connected to Home Assistant via your integration. And I have groups defined in HA (I know I could define them also in EPPSomfy-RTS but I don't want to do that).

Now when I e.g. click on close for the "All" group HA will send 11 `down` commands to ESPSomfy-RTS as fast as it can. Sometimes all 11 commands will be transmitted to the shades but about half of the time it will skip one of these. And by skip I mean that at least the shade does not move but ESPSomfy-RTS thinks it successfully sent the command since it does update the expected shade position.

I think I never saw it skip more than one. Also it's not always the same one that is skipped although most of the time it is one of the first elements in the group. But AFAICT these commands will be sent async so I'm not sure how predictable the order is.

It did work reliably when I was using my prototype ESP32 and only started to have this behavior when miniaturizing it with an ESP32-C3 Super Mini. My first suspect (based on #486) was that the voltage regulator on the C3 could not keep up with the fast successive transmissions leading to a brown out on the transceiver module. Thus I added a 100nF Low-ESR capacitor to the 3.3V rail. That did not help though so this either was not the problem at all or not all of the problem to solve.

It also works reliably when I trigger them one by one (thus implicitly adding a delay between commands).

Now I need some guidance on how to process with debugging this issue. What could I do here? Also: would it be feasable to add a global delay (configurable) between sending commands. I _think_ even a small delay like 10-50ms would already solve my issue.

### How to reproduce it (step by step)

```bash
1. Create group in HA with 10-15 shades
2. Send a command to this group
3. Not all shades actually move while HA "thinks" they all do move
```

rstrouse, 2024-12-16:
The issue is not what it seems.  Every command you sent went out on the radio.  However, not every motor heard it even though it made it to the receiver on the motor.  This is because not all motors recover at the same rate after decoding a frame.  Every frame in the airwaves is decoded by every motor in the listening area.  There is a cooling off period after each frame where most motors can keep up and process the next hardware sync.  However, some are slow to start streaming bytes again.   The solution to slow recovery are repeats.  Simply increase the number of repeats until you get what you desire.  Start with 1 and increase as needed.

I know you do not want to create groups in ESPSomfy RTS but this overcomes the limitations on the motors by reducing the number of frames on the airwaves.  When grouped at the motor, a single frame is sent to command all members but this breaks down when you want to say set all to 54%.  The speed of each motor is going to be different so these are broken into individual frames.

wilriker, 2024-12-16:
Thanks for reminding me of the receiver cool down. That's stuff I knew about eight years ago when I built my first solution on pushstack's blog posts.

But that leaves me with the question: why did it work with not a single one of these "skips" when I ran it with a "normal" ESP32? Was that simply slower in responding to the HTTP requests and thus having a larger delay between the individual commands?

One reason I do not want to create groups is that I do not know how many remotes a motor can store. Currently each motor is paired with four remote addresses already. One of these is obsolete but I cannot unpair it since it died with my previous solution. Do you happen to know how many remotes a motor can store? I also have something in the back my head that this is a ring so newer ones would eventually overwrite the oldest ones. And the first two address are the physical remotes and I don't want to go through the process of running through the house and programming them again. ;-)

rstrouse, 2024-12-16:
The S3 processor and WiFi buffers are faster.  You were getting the benefit of the HTTP request delays adding to the cooldown on the DOWD chip.  The built in cooldown on transmit is optimistic meaning it is set to nominal performance in around the half frame range.  Adding a repeat frame will make sure that it sees the hardware sync followed by the software sync.

Most all motors allow for 12 stored addresses.   However, I have seen more than one motor that was full of abandoned addresses that either came from the factory that way or the installer was goofing around with them before install.  If you run into one you can simply factory reset it and pair it up again.  If you have access to kill the power on it the process is easy.   If not, then it gets a bit more complicated.

wilriker, 2024-12-17:
I went boldly and added each motor in three different groups to ESPSomfy-RTS and it seems that all motors are still happily accepting all physical remotes. :-) I also added a "Repeat 1 Time" to all individual shade controllers (I don't think I will need them for the groups as these are usually not triggered in short succession).

One thing I don't fully understand is what you wrote above:
> When grouped at the motor, a single frame is sent to command all members but this breaks down when you want to say set all to 54%. The speed of each motor is going to be different so these are broken into individual frames.

Does that mean that when I say "Group 'Upstairs' to 54%"
1. it will start the movement for the full group but then will individually stop them at 54% or
2. it will break it down into individual commands  from the start
3. or does "break down" mean "it will not work at all"? ;-)

One last question: instead of adding the repeat frames I would rather see an option to adjust the built in cooldown on transmit. Would that be an option for you to implement? Or point me at the right code location (I was searching myself already but could not find it easily), so I could start implementing it myself and bring up a PR?!

rstrouse, 2024-12-17:
It will break it down into commands for the individual motors.  It has to do this because some may need to go up and some may need to go down to get to 54%.   

You are correct you will likely not need them for the groups but keep in mind it is very difficult to get only one frame sent when pressing the buttons on the remote and you will not notice a repeated frame overall.

wilriker, 2024-12-18:
Yeah, once thinking about it for longer than a second I realized that it could not be No. 1 (except with very complicated code to detect what to use best).

The only thing I realized by now is that the shades commanded to move (from fully open) to an intermediate position actually stop a tad bit late (I happen to have a reference point to compare against) with repeated frames. But that might be a coincidence. Can recheck tonight.

I was again searching for that frame delay and I could not find it for regular commands. The only occurrence I found was when it will be pushed to `tx_queue` but that only happens for frames that were received from a repeater and then will be emitted locally again. This is beginning to drive me crazy. Can you point me to where you add that delay for any regular transmit, please? ;-)

rstrouse, 2024-12-18:
Remember that the transceiver code executes on a single thread.   It is not capable of sending our more than one frame at a time.  So, the cool-down is baked into the sendFrame method as part of the silence on the radio.

https://github.com/rstrouse/ESPSomfy-RTS/blob/eb75868adb92b3e6f5d3ead069f915d84a72efde/Somfy.cpp#L4312

wilriker, 2024-12-19:
Ohhh! I did not realize that you are bit-banging these commands to the radio (I don't know either why I did not find the `sendFrame` method - sloppy research I guess).

That's what pushstack also did with his initial implementation based on SI4010 (also my aforementioned "old solution") but later he switched to (or at least tested with) a SX1231 (RFM69) that had the advantage of pushing the full frame to the transceiver together with instructions on bit-rate (I never got that one running though due to some issues with my RPi and the SPI bus that I never got solved). Up until this point I _thought_ that you do the same with the CC1101 transceiver.

I think I spotted two typos in code comments while looking at the method:
1. https://github.com/rstrouse/ESPSomfy-RTS/blob/eb75868adb92b3e6f5d3ead069f915d84a72efde/Somfy.cpp#L4335 where it says 24 and 14 pulses but looking at the call sites and the comment of the `if` block above this should rather be 12 and 6 respectively, shouldn't it?
2. https://github.com/rstrouse/ESPSomfy-RTS/blob/eb75868adb92b3e6f5d3ead069f915d84a72efde/Somfy.cpp#L4367 where the comment says to end with a 0 but then you actually write out a 1.

rstrouse, 2024-12-19:
The 56 and 80 bit protocols are fundamentally different in their output of the inter-frame silence and the hardware sync.  It is actually 12 and 4 for the 56-bit    Also, with OOK the transition is what is important so pulling high in this case represents a 0 since the toggle is based upon the start of the software sync.

I did try to ship all of it into the buffers for transmission but it worked out better to be able to control the bitstream by bit banging the TX pin.

wilriker, 2024-12-20:
> they act like they paired successfully but they do not actually store the address

Meh! Just realized that this was exactly how two of the motors acted. I will have to go through the process of resetting them then. I have circuit breakers for each single one of them but I am a bit afraid of setting the end positions. How reliable is torque-based auto-finding out the positions?

rstrouse, 2024-12-20:
> ...but I am a bit afraid of setting the end positions. How reliable is torque-based auto-finding of these positions?

Look online you will see plenty of videos for your motor.  In general the endpoint settings are the same for the motors.   I know of not procedure for finding the end positions based on torque, rather it is a matter of nudging the motor to each endpoint.

wilriker, 2024-12-21:
The German FAQ of Somfy on [how to reset the motors](https://www.somfy.de/hilfe/faq?question=wie-kann-ich-einen-funkmotor-neu-programmieren) speaks about three methods of setting the endpoints. Fully manual, fully torque-based or bottom manual and top torque-based.

But that page is pretty bad in terms of missing structure and missing information as it does not say of that is applicable to all motors or just some or whatever else.

 I never did the initial setup. I am not even certain I anywhere have some information on which exact motors have been installed in the house (but I think I should have that somewhere). So I have to rely on the Internet. :-D

011V32, 2024-12-25:
Did you find theyr "RTS-Handbuch"?

wilriker, 2024-12-25:
@011V32 not yet. Haven't looked deeper yet, either. Do you happen to have a link?

wilriker, 2024-12-25:
Since this never actually was a bug I am going to close this now.

closed 2024-12-25 by wilriker