rstrouse/ESPSomfy-RTS #514: Wrong offset for RTW decoding

VolkerMer, 2024-12-29:
### Hardware

ESP32

### Firmware version

since v2.1.4

### Application version

since v2.1.4

### What happened? What did you expect to happen?

The RTW frame decoding gives wrong results, they are offset by 1. 

void somfy_frame_t::decodeFrame(byte* frame) was changed in commit `74ee9f6f25fe5b63138a3e1d037124c5ba5a1095`.
Before: 
```c++
switch(this->encKey) {  
  case 149:
  case 133:
    this->cmd = somfy_commands::My;  
```
...
Then:
```c++
else if(this->encKey > 133) {
   this->proto = radio_proto::RTW;
   this->cmd = (somfy_commands)(this->encKey - 133);
```
As somfy_commands::My == 1, the subtrahend should be 132

### How to reproduce it (step by step)

```bash
1. Go to log window
2. Click on any button of an RTW remote
...
```

### Logs

```bash
{"encKey": 136,
  "address": 7339838,
  "rcode": 2898,
  "command": "My+Up",
  "rssi": -37,
  "bits": 56,
  "proto": 1,
  "valid": true,
  "sync": 4,
  "pulses": [102452,47299,10146,97684,2604,2564,2533,2596,4872,1281,1314,629,660,628,665,1276,1309,644,655,639,656,644,646,1289,648,644,653,636,656,645,644,637,661,624,668,630,1311,1269,681,615,654,643,1300,1291,1294,639,680,618,673,613,683,1273,1305,626,667,1280,657,644,1296,641,649,1269,1336,618,674,1265,1322,634,661,1267,668,634,1311,1268,659,615,1327,1281,657,619,1319,1267,1321,637,658,644,646,1289,1299,633,657,629,671],
  "time": "2024-12-28T17:50:55.385+0100"
},
{"encKey": 133,
  "address": 7339838,
  "rcode": 2899,
  "command": "Unknown(15)",
  "rssi": -42,
  "bits": 56,
  "proto": 1,
  "valid": false,
  "sync": 4,
  "pulses": [70071,32277,102404,102368,99866,2593,175,102356,673,10118,91598,1116,4985,2593,2563,2553,2569,4893,1281,1305,624,682,614,679,617,676,1260,1315,1289,1298,1285,657,642,646,637,661,624,668,631,660,641,1301,630,660,1291,643,633,660,640,1298,1284,1305,1275,1335,620,655,1283,1317,635,658,1270,654,635,1305,649,658,1267,1319,636,656,1269,1333,620,668,1258,664,644,1304,1274,674,613,1310,1270,687,627,1295,1288,1299,632,660,640,674,1272,1311,618,675,639,657],
  "time": "2024-12-28T17:51:00.165+0100"
},
{"encKey": 134,
  "address": 7339838,
  "rcode": 2900,
  "command": "My",
  "rssi": -42,
  "bits": 56,
  "proto": 1,
  "valid": true,
  "sync": 4,
  "pulses": [102400,1109,101262,19864,15995,44000,22577,9484,27918,18916,10128,59637,529,502,650,1371,474,1176,1749,19918,11683,2617,2555,2560,2577,4875,1292,1294,636,661,623,668,631,660,1291,668,609,1310,646,671,1265,672,620,655,639,657,644,1295,642,649,1294,1312,1268,659,641,651,617,1328,624,665,1280,1307,623,682,616,676,1268,1311,643,645,1286,665,621,1306,646,668,1261,1329,625,659,1266,1320,638,657,1271,656,639,1306,1277,659,647,1293,1292,645,649,1309,1270,1304,623,668,657,636,1280,1319,641,651,643,652],
  "time": "2024-12-28T17:51:02.460+0100"
},
{"encKey": 136,
  "address": 7339838,
  "rcode": 2901,
  "command": "My+Up",
  "rssi": -38,
  "bits": 56,
  "proto": 1,
  "valid": true,
  "sync": 4,
  "pulses": [102396,102367,40736,61702,63195,10144,97696,2565,2594,2536,2570,4894,1288,1300,631,660,630,671,1272,1309,643,654,639,655,642,647,1289,649,644,652,639,656,619,1322,639,649,644,654,638,656,1269,683,614,680,617,1323,614,667,1281,656,643,1296,641,648,1295,1311,617,676,1289,645,622,1308,645,669,1258,1306,650,659,1266,1319,639,658,1270,656,640,1316,1271,661,624,1320,1265,672,621,1308,1272,1326,641,643,649,659,1266,1320,614,684,620,670],
  "time": "2024-12-28T17:51:11.174+0100"
}

in time order:
down -> My + up     = 136
My   -> Unknown(15) = 133
up   -> My          = 134
down -> My + up     = 136
```

VolkerMer, 2024-12-29:
Of course also 
  else if(this->encKey > 133) {  
should be changed to > 132 or >= 133

rstrouse, 2024-12-31:
Which motors do you have?  I don't have this equipment and another user tested this and said it was correct.  I'll look at my notes to see if maybe I need another protocol type.

VolkerMer, 2025-01-01:
The motors are asa ETR3 with the elegance RTW remote control

https://asamotor.com/en/linea.php?brand=2&lineaid=328

The ESPSomfy-RTS wiki "Set up the Transceiver" is showing in the log
window the key values for up as I see them, 
but in the current version of the s/w the decoding is different. The
encoding for commanding does it right.
I do not have any problems to command up and down in the start window
(attached) after pairing the motors.
The linking of the remote however does not work. I assume this is due
to the decoding of the key pressed during the link step.

Regards
Volker

P.S. I am not able to compile with Arduino IDE 2.3.3.  and 2.3.4. There
are several libraries for Wifi and networking.
Could you please indicate which ones to use?

[1] view it on GitHub
    https://github.com/rstrouse/ESPSomfy-RTS/issues/514#issuecomment-2566730606
[2] unsubscribe
    https://github.com/

82 |   wifi_event_id_t onEvent(WiFiEventSysCb cbEvent, arduino_event_id_t event = ARDUINO_EVENT_MAX);
      |   ^~~~~~~~~~~~~~~
/home/mertens/.arduino15/packages/esp32/hardware/esp32/3.0.7/libraries/WiFi/src/WiFiGeneric.h:41:25: error: 'NetworkEventCb' has not been declared
   41 | #define WiFiEventCb     NetworkEventCb
      |                         ^~~~~~~~~~~~~~
/home/mertens/.arduino15/packages/esp32/hardware/esp32/3.0.7/libraries/WiFi/src/WiFiGeneric.h:83:20: note: in expansion of macro 'WiFiEventCb'
   83 |   void removeEvent(WiFiEventCb cbEvent, arduino_event_id_t event = ARDUINO_EVENT_MAX);
      |                    ^~~~~~~~~~~
/home/mertens/.arduino15/packages/esp32/hardware/esp32/3.0.7/libraries/WiFi/src/WiFiGeneric.h:83:41: error: 'arduino_event_id_t' has not been declared
   83 |   void removeEvent(WiFiEventCb cbEvent, arduino_event_id_t event = ARDUINO_EVENT_MAX);
      |                                         ^~~~~~~~~~~~~~~~~~
/home/mertens/.arduino15/packages/esp32/hardware/esp32/3.0.7/libraries/WiFi/src/WiFiGeneric.h:43:25: error: 'NetworkEventSysCb' has not been declared
   43 | #define WiFiEventSysCb  NetworkEventSysCb
      |                         ^~~~~~~~~~~~~~~~~
/home/mertens/.arduino15/packages/esp32/hardware/esp32/3.0.7/libraries/WiFi/src/WiFiGeneric.h:84:20: note: in expansion of macro 'WiFiEventSysCb'
   84 |   void removeEvent(WiFiEventSysCb cbEvent, arduino_event_id_t event = ARDUINO_EVENT_MAX);
      |                    ^~~~~~~~~~~~~~
/home/mertens/.arduino15/packages/esp32/hardware/esp32/3.0.7/libraries/WiFi/src/WiFiGeneric.h:84:44: error: 'arduino_event_id_t' has not been declared
   84 |   void removeEvent(WiFiEventSysCb cbEvent, arduino_event_id_t event = ARDUINO_EVENT_MAX);
      |                                            ^~~~~~~~~~~~~~~~~~
/home/mertens/.arduino15/packages/esp32/hardware/esp32/3.0.7/libraries/WiFi/src/WiFiGeneric.h:84:8: error: 'void WiFiGenericClass::removeEvent(int, int)' cannot be overloaded with 'void WiFiGenericClass::removeEvent(int, int)'
   84 |   void removeEvent(WiFiEventSysCb cbEvent, arduino_event_id_t event = ARDUINO_EVENT_MAX);
      |        ^~~~~~~~~~~
/home/mertens/.arduino15/packages/esp32/hardware/esp32/3.0.7/libraries/WiFi/src/WiFiGeneric.h:83:8: note: previous declaration 'void WiFiGenericClass::removeEvent(int, int)'
   83 |   void removeEvent(WiFiEventCb cbEvent, arduino_event_id_t event = ARDUINO_EVENT_MAX);
      |        ^~~~~~~~~~~
/home/mertens/.arduino15/packages/esp32/hardware/esp32/3.0.7/libraries/WiFi/src/WiFiGeneric.h:44:25: error: 'network_event_handle_t' has not been declared
   44 | #define wifi_event_id_t network_event_handle_t
      |                         ^~~~~~~~~~~~~~~~~~~~~~
/home/mertens/.arduino15/packages/esp32/hardware/esp32/3.0.7/libraries/WiFi/src/WiFiGeneric.h:85:20: note: in expansion of macro 'wifi_event_id_t'
   85 |   void removeEvent(wifi_event_id_t id);
      |                    ^~~~~~~~~~~~~~~
/home/mertens/.arduino15/packages/esp32/hardware/esp32/3.0.7/libraries/WiFi/src/WiFiGeneric.h:129:25: error: 'arduino_event_id_t' has not been declared
  129 |   const char *eventName(arduino_event_id_t id);
      |                         ^~~~~~~~~~~~~~~~~~
/home/mertens/.arduino15/packages/esp32/hardware/esp32/3.0.7/libraries/WiFi/src/WiFiGeneric.h:131:30: error: 'arduino_event_t' has not been declared
  131 |   static void _eventCallback(arduino_event_t *event);
      |                              ^~~~~~~~~~~~~~~
/home/mertens/.arduino15/packages/esp32/hardware/esp32/3.0.7/libraries/WiFi/src/WiFiGeneric.h:83:68: error: 'ARDUINO_EVENT_MAX' was not declared in this scope; did you mean 'WIFI_EVENT_MAX'?
   83 |   void removeEvent(WiFiEventCb cbEvent, arduino_event_id_t event = ARDUINO_EVENT_MAX);
      |                                                                    ^~~~~~~~~~~~~~~~~
      |                                                                    WIFI_EVENT_MAX
/home/mertens/.arduino15/packages/esp32/hardware/esp32/3.0.7/libraries/WiFi/src/WiFiGeneric.h:84:71: error: 'ARDUINO_EVENT_MAX' was not declared in this scope; did you mean 'WIFI_EVENT_MAX'?
   84 |   void removeEvent(WiFiEventSysCb cbEvent, arduino_event_id_t event = ARDUINO_EVENT_MAX);
      |                                                                       ^~~~~~~~~~~~~~~~~
[... 659 lines cut ...]
In file included from /home/mertens/.arduino15/packages/esp32/hardware/esp32/3.0.7/libraries/WiFi/src/WiFiSTA.h:29,
                 from /home/mertens/.arduino15/packages/esp32/hardware/esp32/3.0.7/libraries/WiFi/src/WiFi.h:33,
[... cut ...]
/home/mertens/.arduino15/packages/esp32/hardware/esp32/3.0.7/libraries/WiFi/src/WiFiGeneric.h:44:25: error: 'network_event_handle_t' does not name a type; did you mean 'network_prov_event_handler_t'?
   44 | #define wifi_event_id_t network_event_handle_t
      |                         ^~~~~~~~~~~~~~~~~~~~~~
/home/mertens/.arduino15/packages/esp32/hardware/esp32/3.0.7/libraries/WiFi/src/WiFiGeneric.h:82:3: note: in expansion of macro 'wifi_event_id_t'
   82 |   wifi_event_id_t onEvent(WiFiEventSysCb cbEvent, arduino_event_id_t event = ARDUINO_EVENT_MAX);
      |   ^~~~~~~~~~~~~~~
/home/mertens/.arduino15/packages/esp32/hardw

rstrouse, 2025-01-01:
You must use Arduino Core 2.0.17 to compile the code.  Arduino Core 3.0.7 code segments are too big to fit on the flash.  I think the Wifi 6 code added almost 200k to it and you cannot def it out.

[image]

VolkerMer, 2025-02-04:
I installed Arduino IDE 1.8 and compiled the project with the changes I suggested. This then was installed on a second system.
The log and the screenshot of the chnaged version v2.4.7 **V**2.4.7 show, that the coding of the commands is different to the decoding (previous log).  
  
```
[{"encKey": 136,
  "address": 504825,
  "rcode": 68,
  "command": "Down",
  "rssi": -26,
  "bits": 56,
  "proto": 1,
  "valid": true,
  "sync": 4,
  "pulses": [10214,7365,2601,2525,2564,2557,4881,1260,1281,620,676,608,687,1244,1304,614,654,618,694,590,686,1250,666,612,687,587,690,594,670,610,660,617,1304,626,654,1263,648,644,656,614,684,592,669,607,682,630,1291,605,680,632,639,1243,682,613,665,623,1305,1240,1323,613,668,597,693,1236,689,595,670,605,657,639,1319,1236,682,591,1323,609,687,598,668,1256,662,597,699,577,1331,598,691,1252,667,596,672,602,1332,1249,642,632,686,588],
  "time": "2025-02-04T11:29:43.199+0100"
},
{"encKey": 133,
  "address": 504825,
  "rcode": 69,
  "command": "My",
  "rssi": -27,
  "bits": 56,
  "proto": 1,
  "valid": true,
  "sync": 4,
  "pulses": [10247,7328,2587,2507,2607,2525,4870,1251,1318,613,691,619,645,629,659,1270,1296,1264,1286,1270,657,615,678,617,647,636,662,591,697,576,702,607,1294,1269,684,575,693,593,672,637,636,642,651,616,674,615,1292,620,668,1264,639,646,670,586,1306,1275,1301,589,689,621,668,1261,664,620,669,612,686,578,1330,1258,661,615,1305,601,659,638,657,1269,651,611,684,595,1297,632,687,1243,649,611,687,600,1315,1280,639,625,671,605],
  "time": "2025-02-04T11:29:44.992+0100"
},
{"encKey": 134,
  "address": 504825,
  "rcode": 70,
  "command": "Up",
  "rssi": -26,
  "bits": 56,
  "proto": 1,
  "valid": true,
  "sync": 4,
  "pulses": [10224,7338,2600,2545,2566,2528,4888,1265,1299,609,685,625,644,624,680,1241,664,618,1303,612,666,1270,655,635,633,641,669,609,656,605,1316,622,652,613,665,1272,673,587,686,624,640,626,658,639,1282,621,650,637,690,596,662,1250,685,599,665,599,1327,1247,1319,617,654,619,647,1287,652,622,643,632,663,626,1280,1277,660,627,1294,634,636,642,649,1272,632,638,656,618,1321,613,655,1270,657,612,662,621,1297,1284,640,644,653,613],
  "time": "2025-02-04T11:29:46.434+0100"
}]
```  

[image]

Seresam, 2025-08-06:
Hello,
I think I am experiencing the same issue: commands work fine but   commands from physical remotes displayed in the logs don't match the actual commands. A minor drawback is that the shade position is not updated when operating the shades with physical remotes.