rstrouse/ESPSomfy-RTS #517: Consecutive commands to two different motors are registered, but RTS command is sent only to the first one

smumriak, 2025-01-01:
### Hardware

ESP32

### Firmware version

v2.4.7

### Application version

v2.4.7

### What happened? What did you expect to happen?

Closing one shade from home assistant and immediately closing another one. Only the first shade started moving. Status is reported as if both are moving. You can reproduce this from web interface too, but you have to click the second one veeeeery fast, like 200ms fast. This, unfortunately, break any possible home assistant automations that involve more than one motor and creating groups for them is not the best workaround (I have three motors, so it's 3 shades + 4 groups to cover all possible cases).

### How to reproduce it (step by step)

```bash
1. Set up two motors
2. Send any open/close command to both of them in a very short time frame.
3. Observer

In the attached logs I also can see that first command indeed has "Sending Shade Command" log line, while the second does not.
```

### Logs

```bash
Sending Shade Command
Received:{"shadeId":1,"command":"down"}
CMD:Down ADDR:9901383 RCODE:8682 REPEAT:0
Enabled receive on Pin #12 Timing: 0
                                    Timing WebServer: 176ms
                                                           Sending Shade Command
Received:{"shadeId":2,"command":"down"}
CMD:Down ADDR:1590819 RCODE:7200 REPEAT:0
Enabled receive on Pin #12 Timing: 0
                                    Timing WebServer: 175ms
```

smumriak, 2025-01-01:
I find it much easier to reproduce via simple shell command
```bash
for shade in 1 2; do mosquitto_pub -h mqtt -u "user" -P "password" -t shades/$shade/target/set -m 100; done
```

rstrouse, 2025-01-01:
This issue has to do with the packet recovery on the motor.  Every motor processes every radio frame it hears.  However, some motors are faster at recovering to receive the next frame than others.   To fix this issue add a couple of repeats to the tardy motors in the shade configuration.  This will ensure the motors will hear the frame if they are cooling off from the last one.

smumriak, 2025-01-01:
You are right, thank you! Added two repeats to each and it now works perfectly.

closed 2025-01-01 by smumriak

poudenes, 2025-05-12:
Hi All,

I created a general question: https://github.com/rstrouse/ESPSomfy-RTS/issues/570

To add repeats will this solve my question also?