rstrouse/ESPSomfy-RTS #519: Ok I messed up

1mfaasj, 2025-01-02:
### Hardware

ESP32

### Firmware version

v2.4.6

### Application version

v2.4.6

### What happened? What did you expect to happen?

Ok I messed up and need some info to fix this as good as possible. Because of unfamiliarity at the start and playing with your solution along the way, the memory of some shutters are full and some of them are not configured correctly (for instance when I close one shutter, 2 other shutters responds, don't ask me why)

**My setup:**
I use the somfy connexoon rts device (the hub) and an ESP with your software on it.
and all shutters (5 in total) are paired with the two in some way.

**What I want to achieve:** 
only use the ESP and your solution because of the additional possibilities.

**What do you suggest how to proceed?**
I know that a reset is a option but unfortunately I can't get to the power supply of the shutters and I need a mechanic for it. I'm  looking for something I can do myself in the meantime.

Is it correct that when I unpair the shutters from the somfy hub, this creates space in memory?
The shutters in your solution aren't correct configured either, so I think that I need to unpair all of them first, or not? I cannot find the unpair button btw. Can you please explain how to do this?

Thanks

### How to reproduce it (step by step)

```bash
see description above
```

### Logs

```bash
no logs
```

rstrouse, 2025-01-02:
> Is it correct that when I unpair the shutters from the somfy hub, this creates some space in memory? my first thought, when I inpair the shutters from the somfyhub and the esp, and start over; it should have enough memory to pair them all again right?

Yes it is true.  In fact if you have a channel from a remote (ESPSomfy RTS, Connexoon, handheld, ...etc) that is controlling a motor that you do not want it to control simply unpair it and it will stop doing that.

> The shutters in your solution aren't correct configured either, so I think that I need to unpair all of them first, or not? I cannot find the unpair button btw. Can you please explain how to do this?

If you try to pair an already paired motor it will unpair.   Just click the button that the paring was not successful and it will show that it is in an unpaired state.

> Do you know how many times you can pair shutters? what is the capacity of the memory?

Most motors have a capacity of 12 but it is not uncommon for installers to set the motors up in their shop prior to installation and consume some of these slots.  They may also do this at the factory.

ALSO:  Whatever you do, do not reuse an address from your remotes or Connexoon in ESPSomfy RTS.  If you see an address in the log please resist the temptation to use it.

1mfaasj, 2025-01-03:
thanks for the reply!
So when I want to unpair a shutter, I just need to go to Shades and click on Pair Shade again?
[image]
I see also a Prog button (when I go to Virtual Remote), this is not what you mean right?

and can you clarify a bit what you mean with reuse an address? How do I know?
You mean the name of the shade/shutter or?
I also see this numbers here? But if you mean these numbers, I don't have control over it right?
[image]

rstrouse, 2025-01-03:
You can use the virtual remote or the pairing button to unpair the motors either one will work.  In the end you just need to send the prog frame to the motor when its memory is open.  That being said you cannot unpair a remote that opened the memory on the motor.  So open the memory with a handheld remote and then unpair ESPSomfy RTS.

When I say do not share the addresses I am referring to the remote address in the shade configuration.  Unfortunately, some folks have assumed that they need to replicate a working remote by using the address they see in the log when setting it up.  This almost always leads to factory reset.

1mfaasj, 2025-01-17:
after I pressed the pair button on the handheld remote and clicked on Pair shade on the already configured shade in ESPSomfyRTS; do I still need to remove the shade in ESPSomfy RTS? with these buttons?
[image]

but what about this bin?
[image]

After I did the action as described above, and pressed the ^ button, the shade still responded and goes up ?!
the shade should be unpaired now right?

1mfaasj, 2025-01-18:
Ok another strange thing I found out

The handheld remotes linked to each shade are still correct, I would expect when the shades are not responding correctly the linked remotes isnt either.

```
last_cmd: Up
cmd_source: remote
cmd_address: 10195300
cmd_fired: 1737205021.622573
```
So when I press Up on the handheld remote of shade 1, in home assistant the attributes become visible for the same/correct shade.

rstrouse, 2025-01-20:
Your primary issue was that the motor memory was still open when you paired the next ESPSomfy RTS shade.  Depending on the motor you have to wait until the motor closes or you manually need to close it after you have paired a remote.  You can simply unpair the ESPSomfy RTS setting from the motor that is not supposed to be controlled by it.

Unfortunately, the same indication is given for when the memory is opened as when it is closed.  To add to the confusion it is the same indication as when a remote is paired as well as unpaired.  You need to make sure you keep a mental picture as to how many times the motor has jogged.

So for some motors you long press on the prog button to open the memory.   Then you pair the remote with a short press.   Then after the motor jogs you close the memory with another short press on the prog button from the remote that originally opened the memory.   Somfy must deal with this a lot as they recommend removing power from the motors you do not want to pair.

sasikoa, 2025-07-31:
Hi all, sorry to revive this thread but I am having issues in the pairing process. 
I had it working for some months but I changed my Internet company and was not able to connect the esp32 to the new network, so I decided to reset and start the process again.
The problem is when I arrive to the pairing step, I open the memory with the remote but now it is not pairing with the prog button. Must I assume that as I didn't unpair before reseting the previous shade, now the memory of the motor is full and cannot pair it again?
How could I solve it? I am a completely rookie on this and don't want to make a mess and erase the remote (for example).

Thanks in advance!