rstrouse/ESPSomfy-RTS #522: Device goes offline

Musaban1, 2025-01-05:
Hi! I set up an ESP32 this summer with FW 2.4.6, and it worked perfectly until a couple of months ago when it became unavailable. I tried power cycling it, and it would then stay available for a maximum of 24 hours before becoming unavailable again. I have now tried switching to a newer ESP32 and updating to FW 2.4.7, but I’m still experiencing the same issue. I don’t know what else to try. Any advice?

rstrouse, 2025-01-06:
Are you running Home Assistant?  If you are you should be able to see what the history of the memory and network strength has been.  Try turning off Roaming under network and broadcast UPnP over SSDP from the system page.

Musaban1, 2025-01-06:
Here is a picture of the statistics after the ESP went offline; I can't see anything unusual.
I have tried turning off roaming and broadcasting UPnP over SSDP now, so we'll see how long it stays online.

[image]
[image]

rstrouse, 2025-01-06:
Also turn off Auto Check for updates.  Your memory and Wifi strength looks good.   It does check once a day to see if there are any posted updates.

Musaban1, 2025-01-08:
Now it has been online for two days, so this looks promising! Thanks for your help.

albebert, 2025-01-20:
seems that i've same issue. 

disabling roaming seems to work, so i think that there is a problem with roaming.

i've a good wifi coverage with severals access point. 

i actually do tests on my desk, with an AP in the room, other ap that the esp can find are not on the same channel to prevent interference.

i can send debug log if exists. (i'm now testing esp somfy rts in order tu replace my old rflink)

rstrouse, 2025-01-20:
Are you running a mesh or dedicated access points?

albebert, 2025-01-20:
dedicated Fortinet AP, managed by a fortigate (enterprise solution).