rstrouse/ESPSomfy-RTS #523: Suggested pins for ESP32-C3-DevKitM-1

jiwjiwjiw, 2025-01-07:
Hi,

Thanks a lot for this great project that works beautifully. I can think of a lot of commercial projects whose documentation is not that good, and lacks your very enjoyable sense of humor.

In case you want to extend the list of [suggested pins](https://github.com/rstrouse/ESPSomfy-RTS/wiki/Simple-ESPSomfy-RTS-device#suggested-pins), here is a mapping that works perfectly with the ESP32-C3-DevKitM-1 board (pin reference for the E07-M1101D) : 

Pin | Description | ESP Pin
-- | -- | --
1 | GND | GND
2 | VCC | 3v3
3 | GDO0 - This is the TX Pin | GPIO 4
4 | CSN | GPIO 5
5 | SCK | GPIO 6
6 | MOSI | GPIO 7
7 | MISO | GPIO 2
8 | GDO2 - This is the RX pin | GPIO 3

rstrouse, 2025-01-09:
Excellent.  Thank you I will update the documentation soon.

mAPBhlJ, 2026-10-19:
@jiwjiwjiw you may want to add an image of the board showing the labes such as in [my issue](https://github.com/rstrouse/ESPSomfy-RTS/issues/694)

Personally, I think it is much clearer, or are the GPIO numbers the same as the numbers on the board?
In case of the Xiao ESP32-C3 for example, the pin numbers are not the same as GPIO numbers.