rstrouse/ESPSomfy-RTS #524: Control through USB

jayhoward1234, 2025-01-09:
Hello and thank you for your work.
I have a use case where I would like to control the blinds from a Windows PC with no WiFi available.
Is it possible to control the ESPSomfy-RTS device locally through USB serial commands?
Thanks!

rstrouse, 2025-01-09:
Unfortunately, the command interface is not connected to the serial port.  The commands are routed through a network socket.  It will stand up with a dedicated AP though.

011V32, 2025-01-15:
> Hello and thank you for your work. I have a use case where I would like to control the blinds from a Windows PC with no WiFi available. Is it possible to control the ESPSomfy-RTS device locally through USB serial commands? Thanks!

Probably you are able to use Ethernet/LAN instead of WiFi?

jayhoward1234, 2025-01-15:
Thanks for the suggestion. I ordered an ethernet model today to try out.

rstrouse, 2025-01-15:
Adding the code should be fairly straightforward if you have a coding background.  You may run out of codespace if you add libraries though.  For your own build you could repartition to eliminate the OTA updates.

jayhoward1234, 2025-01-15:
Ok, I have two good options for moving forward. Thanks for the help!