rstrouse/ESPSomfy-RTS #525: Need Troubleshooting Suggestions - No RSSI

scott1m, 2025-01-17:
Hello,

Like others have stated, I’m loving this integration and instructions.  This is my first ESP32 project, so maybe I’m missing something obvious.

I purchased a WROOM ESP32 and a “generic” CC1101 Transceiver - but it does appear similar to the one pictured in the wiki.

I’m at the point where I click the button on my RTS remote, but I do not see any signal reception or logs.  I’ve checked the wiring a couple of times.  Occasionally I get the “Radio Not Configured” message when broadening the spectrum, but often it goes away.  I’ve included a few pictures that maybe someone can point out where I’ve went wrong.

Thanks in advance.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

rstrouse, 2025-01-17:
You must check the Enable Radio checkbox then press Save Radio.

scott1m, 2025-01-17:
That was bad timing on my screen grab.  I do have the checked.  The “Radio not initialized” appears, but if a save a time or two more, it goes away.  Still no signal.

[image]

rstrouse, 2025-01-20:
Increase your RX bandwidth by 50 at a time to see if you start getting signal.  The bandwidth tuning on the CC1101 is tighter than the bandwidth from the remote.   Also, please remove your antenna and verify that it is not an RP-SMA.  It should have a male pin inside it.

scott1m, 2025-01-20:
Still no luck.  I increased the bandwidth to ~145 then wider and wider.  Also, verified that it does have a male pin.

[image]
[image]

scott1m, 2025-01-24:
Given that this is not working, would you recommend purchasing a new CC1101 transceiver?

RobertoBiundo, 2025-04-18:
Did you ever fix this. I have the same issue with the same controller. I'm really stuck. Have tried everything. Even changed all the GPIO pins to see if one of them was causing issues

scott1m, 2025-04-18:
I never got this working either. Still hoping to find the fix. At least I’m not alone.

RobertoBiundo, 2025-04-19:
So I figured it out last night going over all the posts and turns out that the model 5 we have is an IO Home one and it does not have RTS protocol. 

Since I still want the automation I found a cheap controller on the used market and will try buying it. Disassemble the controller and solver a couple of wires on the buttons to control them directly with the ESP32. Turn the controller into some sort of daughter board and 3d print some box for the combo. 
I also disassembled mine yesterday. There are test points that you can easily solder on. It uses a 3V battery, so you might be able to even power it with the ESP32 itself. 

Do you care about updates on this. I will probably create my own code and skip the ESPRTS all together since I love tinkering with my own stuff

scott1m, 2025-04-24:
Yes, please let me know if it works out.

sanmagal, 2025-09-07:
My setup was working fine. All of a sudden I GET no RSSI and no entries on the logs. Tried rebooting with no success and tuning also without success. 
Any help?