rstrouse/ESPSomfy-RTS #527: Sunny and Windy flags on the display ?

carminou, 2025-01-28:
Hello,

first of all, many thanks for this piece of software and documentation!

I installed and configured ESPSomfy-RTS a few days ago. I have around 20 SIMU (not Somfy) roller shades working with the RTS protocol and all was running smoothly.

Then some decorations appeared in the interface for two, the today five, maybe more tomorrow?) shades: a small sun and/of a small warning sign on the left:

[image]

I suppose these are sunny and windy indicators, but I do not understand why they appear. I have no detector. These shades are receiving only, they do not send any status frames... 

Any idea of what is going on ?

carminou, 2025-02-03:
up up
Please any answer ?

carminou, 2025-02-03:
Answering to myself...
Why did this happened ? I do not know. Maybe a glitch in radio transmission ?

How to resolve ? Using the virtual remote: go in the parameters, click Virtual Remote, select the shade with and unwanted sunny or windy mark, be sure that the Sunny an Windy check-boxes at the bottom are unchecked and click send.

Nickduino, 2025-02-19:
[image]

 ?