rstrouse/ESPSomfy-RTS #529: Keygo io control

Blastmun, 2025-01-31:
### Describe you new feature you'd like

I haven't found anything on the internet about wiring a keygo IO remote control on esp32 to control it, do you know if this is possible and if so how to wire it correctly?

Blastmun, 2025-02-02:
By performing tests, I notice that on the remote control, one of the poles of the switch is connected to GND. So I soldered cables on the opposite side of the switch. There I encounter very strange problems, I have the impression that between the configured PIN and the activated real pin it does not correspond.
The remote control controls a sliding gate via keygo IO remote control.
For the configuration:
[image]
After doing some tests I have the impression that when I program GPIO14 it is in fact 26 which is triggered and when I program 27 it triggers if I connect to GPIO 14. I have the impression that the "my" button never triggers even if I test numerous PINs.

My board is a esp32 wroomer 30pin and I used the bin esp32.
Anyone have any idea?

Nickduino, 2025-02-19:
This repo is meant for the RTS protocol, it does not work with io-homecontrol devices

Blastmun, 2025-02-27:
Why people do not read the posts correctly .... I know that the project is compatible with RTS, but in my case present I use the functionality that allows a button support on a physical remote control. And in my concerns reassembled according to the type of button chosen and gate, it does not work. If I make believe that I am on a roller shutter with Up/Down it works, if I put Sliding Portal with the same programming Up/Down nothing is going on the ESP side. I assume an error in the code.