rstrouse/ESPSomfy-RTS #530: loss of control of blinds

ApriliaEdd, 2025-02-03:
### Hardware

Olimex ESP32-PoE/EVB

### Firmware version

v2.4.6

### Application version

v2.4.6

### What happened? What did you expect to happen?

Every now and again I get no response from the blinds.  I can ping the ESPSomfy RTS device, I can still access the web interface, I can reboot the ESP from home assistant and from the web interface but once it restarts it is in the same situation.

No requests to the blinds seem to be transmitted either from home assistant or the web interface.  If I power cycle the POE it will return to normal and work correctly. 

I cannot seem to find any reason for this to happen it just seems to stop working.
While it is currently not working is there anything you would like me to try to help diagnose this?

### How to reproduce it (step by step)

```bash
only way to reproduce seems to be to wait...
```

### Logs

```bash

```

yazend, 2025-02-19:
I've been getting a similar behavior lately,
what I've noticed now is that when I try to scan the frequencies at this time, "no radio" is available.
I'm going to start tests with new hardware.
It could be that the "E07-M1101D-SMA" has a problem, what will definitely be taken into account in the next build are heat sinks, even if the devices have a wide temperature tolerance, I am of the opinion that one should not operate this permanently without cooling.

jeff916, 2025-02-21:
I just had the same problem. When I look on the web interface I noticed "Radio not initialized". This was the same even after several reboots. All works after a power cycle. Firmware v2.4.6

A new diagnostic sensor for the radio status would be nice.

yazend, 2025-02-22:
At first a power cycle was also the only way to get the functions back, but now it only works sporadically.
So the new parts have been ordered, as soon as they arrive I will start building and testing.

sco0ts, 2025-09-16:
> wer cycle was also the only way to get the functions back, but now it only works sporadically.
> So the new parts have been ordered, as soon as they arrive I will start building and testing.

Did you ever get this working? If so, what parts did you use (links to amazon please!)?