rstrouse/ESPSomfy-RTS #532: Zigbee support

TB31388, 2025-02-04:
### Describe you new feature you'd like

Have you considered adding support for zigbee? Both to control the upcoming Somfy devices that are zigbee based and to allow the esp32 to integrate into an existing zigbee network, instead of relying on LAN.

jonathan-gatard, 2025-03-13:
+1
It could be interesting to use ESP32 H2

meiser79, 2025-05-01:
check #303

abaghbani, 2025-10-31:
I just measured the power consumption of current project with WLAN on "esp32-s3" and result is around 700mW, and then I just thinking about replacing wlan to zigbee for HA users, of course non HA user can benefit a lot with wlan and using phone only but this consumpton for HA user is a lot and adding zigbee and then turn off wlan radio of esp32 can save power a lot, let's think more about adding zigbee to this nice project