rstrouse/ESPSomfy-RTS #535: not responding after some times

albebert, 2025-02-17:
### Hardware

ESP32

### Firmware version

v2.4.7

### Application version

v2.4.7

### What happened? What did you expect to happen?

The device become unresponsive after few minutes or hours.

The device respond to ping but UI is not reachable and control from HA are not working anymore.

### How to reproduce it (step by step)

```bash
just wait :(
```

### Logs

```bash
logs output from serial :

20:34:53.771 > Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
20:34:53.776 >
20:34:53.776 > Core  0 register dump:
20:34:53.778 > PC      : 0x40127ed8  PS      : 0x00060030  A0      : 0x80135c91  A1      : 0x3ffd7100  
20:34:53.788 > A2      : 0x00000000  A3      : 0x00000038  A4      : 0x00000041  A5      : 0x00000000  
20:34:53.796 > A6      : 0x3ffd5254  A7      : 0x00000000  A8      : 0x801280fb  A9      : 0x3ffd7210
20:34:53.801 > A10     : 0x00000000  A11     : 0x00000038  A12     : 0x00000041  A13     : 0x400871b4
20:34:53.809 > A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001d  
20:34:53.818 > EXCVADDR: 0x80135c91  LBEG    : 0x4008b8ad  LEND    : 0x4008b8cf  LCOUNT  : 0xffffffff  
20:34:53.827 >
20:34:53.827 >
20:34:53.827 > Backtrace: 0x40127ed5:0x3ffd7100 0x40135c8e:0x3ffd7120 0x401af9ab:0x3ffd7140 0x40122e4d:0x3ffd7160 0x401b52fd:0x3ffd7180 0x401b53f9:0x3ffd71d0 0x401b7281:0x3ffd71f0 0x401b4050:0x3ffd7210
20:34:53.843 >
20:34:53.843 >
20:34:53.843 >
20:34:53.843 >
20:34:53.843 > ELF file SHA256: 060a3bd7648b16fe
20:34:53.847 >
20:34:54.062 > Rebooting...
20:34:54.062 > ets Jul 29 2019 12:21:46
20:34:54.065 >
20:34:54.065 > rst:0xc (SW_CPU_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
20:34:54.070 > configsip: 0, SPIWP:0xee
20:34:54.070 > clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
20:34:54.079 > mode:DIO, clock div:1
20:34:54.079 > load:0x3fff0030,len:1344
20:34:54.082 > load:0x40078000,len:13964
20:34:54.084 > load:0x40080400,len:3600
20:34:54.088 > entry 0x400805f0
20:34:54.512 > E (300) psram: PSRAM ID read error: 0xffffffff
```

LaurentN1305, 2025-04-23:
J'ai le même problème, le ping est OK, mais l'interface graphique ne s'affiche pas/

maaraneasi, 2025-04-27:
I am having the same issue - after some time, the esp board stops responding. I was not able to capture any error but I can't reach the web UI nor control any shades. For some reason, the issue got progressively worse and worse.

sco0ts, 2025-09-16:
> I am having the same issue - after some time, the esp board stops responding. I was not able to capture any error but I can't reach the web UI nor control any shades. For some reason, the issue got progressively worse and worse.

did you ever solve this?  did you try and older versions?