rstrouse/ESPSomfy-RTS #541: Waveshare ESP32-S3 ETH POE

Spokplacenta, 2025-03-04:
### Hardware

ESP32-S3

### Firmware version

v2.4.6

### Application version

v2.4.6

### What happened? What did you expect to happen?

Hello, Thank you for your job, this project is great! I installed the firmware on an ESP32-S3 ETH, I manage to control my curtains in RTS and my shutter with a remoteio. 

I only have the ethernet part that resists, I tried several configuration but I can't make Ethernet work.

I don't find W5500 in the chipList : 

[image]

[image]

### How to reproduce it (step by step)

```bash
1. Go to...
2. Click on...
...
```

### Logs

```bash

```

moshiko2312, 2025-08-19:
@Spokplacenta  witch bin file do you write? 
i trying to flash  SomfyController.ino.esp32s3.bin & SomfyController.onboard.esp32s3.bin
but its not going to Wifi

moshiko2312, 2025-08-20:
eys me too 

but do know when is add to the code ?

moshiko2312, 2025-08-20:
@BritInAustria 
hi im using this config

[image]

[image]
[image]
[image]
[image]

moshiko2312, 2025-08-24:
@rstrouse  ?

Spokplacenta, 2026-01-28:
@BritInAustria You can already try my feature on my fork, I added a tutorial to explain how to upload the firmware to the ESP.
I just added some fixes just now to improve stability.

davidyang, 2026-03-03:
[compiled.zip](https://github.com/user-attachments/files/25726457/compiled.zip)

I've attached a built version of the binaries (I did it with instructions from @Spokplacenta's repo).  I'm using the esp32s3.bin and the littlefs included with the Waveshare POE I have and it seems to be working.

Thanks everyone for this thread.  If any of you have a cool way of holding the Waveshare and CC1101 Transceiver would love ideas.

Speedyomi, 2026-03-08:
> > [@Mavyre](https://github.com/Mavyre)
> > i am trying since 2 days to flash the Software with the PlatformIO but i could not did it maybe i am stupid. Maybe you could Help me?
> 
> I don't really know what you did (and thus what you didn't do, so I can't know how to help you 😅

So i download the files from the Repo, install the VSC, install the PlatfromIO Plguin, open the Project and connect the ESP (COM6) and try Method 1 from the Manual,

 Via PlatformIO Toolbar
Click on the PlatformIO icon in the left sidebar
In PROJECT TASKS → esp32-s3-devkitc-1
Click Upload (or Build and Upload)

I got the following errors

[image]

[image]

davidyang, 2026-10-21:
Hey guys - sharing a box I made in nurb.dev for this case and radio - it's not great but wanted to share!

[waveshare_assembled-stl.zip](https://github.com/user-attachments/files/33789664/waveshare_assembled-stl.zip)

[image]