rstrouse/ESPSomfy-RTS #544: Help on "Radio Not Initialized"

barthel-eu, 2025-03-24:
From the recommendations in the [Simple Hardware Guide](https://github.com/rstrouse/ESPSomfy-RTS/wiki/Simple-ESPSomfy-RTS-device) I purchased the recommended [ESP32-Wroom-32](https://www.amazon.de/dp/B0DGG7LXMF) and [CC1101](https://www.amazon.de/dp/B0CPY5LWFY) modules and connected them as described.

While the ESP32 connects, I can't get the CC1101 to be recognized. I read to reverse GPIO12+13, but that did not help either. Any idea, how I could possibly find out if it's a hardware or software issue? Being my first step into ESP32-world, I'm worried it's a PEBKAS 🫣 Or it could be the hardware? Or does it require extra power (doesn't say anywhere)? It's on my Laptop USB for the setup.

I've connected (pictures avail):
|CC1101 | ESP32 | Cable Color|
|--------|--------|--------|
|1 | GND | black|
|2 | 3x3 | brown|
|3 | D13 | red|
|4 | D05 | orange|
|5 | D18 | yellow|
|6 | D23 | green|
|7 | D19 | blue|
|8 | D12 | white|

I double-checked (many times) the CC1101-layout from the [Amazon product page](https://www.amazon.de/dp/B0DGG7LXMF) and rewired trice. Can't find the mistake. Would there be a 101 on troubleshooting the hardware?

Jezza34000, 2025-03-28:
Hi,
What Somfy hardware do you have ? Is it written IO on the remote ?

barthel-eu, 2025-03-28:
It's written RTS on the remote. To be exact TELIS - RTS PURE - 5012441A106 😎
The homekit connected seamlessly, just didn't take 50% of the commands (likely even less). So I decided to do my first ESP32-kit (have a ESP32 powermeter, but that came ready for use).
As mentioned, I have meanwhile reordered a new CC1101 from different maker, hoping that'll work then - due in Tuesday Amazon says, seemingly shipped around greater Europe 🫣 Don't want to know it's CO2 footprint 🤔

barthel-eu, 2025-03-31:
I received the [CC1101 replacement](https://www.amazon.de/dp/B07YX92NMP). After I connected it (same PIN layout) it worked like a charm. Sure still would think it would make sense to have more info how to identify a non-working module beyond "RADIO NOT INITIALIZED" 🫣 IMHO.

closed 2025-03-31 by barthel-eu