rstrouse/ESPSomfy-RTS #561: Somehow I bricked my shade

kenarsuleyman, 2025-04-26:
### Hardware

ESP32

### Firmware version

2.4.7

### Application version

2.4.7

### What happened? What did you expect to happen?

Shade stopped working, I powered it of for hours, and tried to sync original remote but it is not responding to anything, it seems totally dead. Usually after power of, when i click up/down buttons in remote, shade was doing a small up/down moment that indicating it is working now nothing is happening. what should i do? Is it possible that somehow I damage the motors?

### How to reproduce it (step by step)

```bash
This is my first time trying to install it, I followed tutorial to prepare hardware, install firmware etc. everything seemed to fine and I started to adding my shades to ESP, I checked my remote controller button pressed being shown on radio log, I used that address when adding shade, when i click up/down buttons, motor was running but really momentarily, In real remote when i hold a key, motors were working until i stop pressing, but when i was trying to add shade to ESP, even if i keep pressed, motors moved just a small moment of time unless I press again. I tried to pair shade with ESP and I was trying to figure it out what should I do and randomly pressing up/down buttons and suddenly shade stopped working.
```

### Logs

```bash

```

sk-ilya, 2025-05-01:
It's possible that the shade became out of sync with the remote. Have you tried [hard resetting](https://www.somfysystems.com/en-us/support/faq?question=how-to-reset-your-rolling-shutter-motor-to-factory-mode)?

kenarsuleyman, 2025-05-01:
I tried that but didn't work. My only assumption based on my electronics knowledge is that there must be a H bridge inside motor controller and if you pass at the same time forward and reverse commands, it shorts the h bridge and kills mosfets. Maybe something similar to this happened but thankfully the company I purchased my shades accepted to change motor under warranty. When it arrives I will probably buy tahoma switch and don't risk it againGalaxy cihazımdan gönderildi
-------- Orijinal mesaj --------Kimden: Ilya ***@***.***> Tarih: 1.05.2025  14:20  (GMT+03:00) Alıcı: rstrouse/ESPSomfy-RTS ***@***.***> Cc: Süleyman KENAR ***@***.***>, Author ***@***.***> Konu: Re: [rstrouse/ESPSomfy-RTS] Somehow I bricked my shade (Issue #561) 
sk-ilya left a comment (rstrouse/ESPSomfy-RTS#561)
It's possible that the shade became out of sync with the remote. Have you tried hard resetting?

kenarsuleyman, 2025-05-01:
> It's possible that the shade became out of sync with the remote. Have you tried [hard resetting](https://www.somfysystems.com/en-us/support/faq?question=how-to-reset-your-rolling-shutter-motor-to-factory-mode)?

I tried that, but it didn’t work. Based on my electronics knowledge, I assume there must be an H-bridge inside the motor controller. If you send both forward and reverse commands at the same time, it likely shorts the H-bridge and damages the MOSFETs. Maybe something like that happened, but thankfully the company I purchased my shades from agreed to replace the motor under warranty. When it arrives, I’ll probably buy a Tahoma switch and avoid taking the risk again.