rstrouse/ESPSomfy-RTS #563: I can get signals from control but can’t send to motors

GabrielSimoes1983, 2025-04-27:
Hello

I have just set up a device using Esp32 S3 and cc1101

Esp32:
https://a.aliexpress.com/_mtf2H2x

Cc1101:
https://a.aliexpress.com/_mPpAzdp

Everything is working until pairing.
I can enable radio, log Tellis 4 control commands and even pair the control with a roller share in ESPSomfy.
But when I put the motor in pairing mode and try to pair nothing happens. As if I can’t send signals.

I have tuned the radio using the signal check and parameters seem fibe (433.40) rts, …

How can I troubleshoot what is going on?
The only thing I also see is the device rebooting from time to time….

If necessary I can send pictures.

GabrielSimoes1983, 2025-04-29:
I am constantly also seeing the system abort and ESP reboot on some error I can´t identify:

[22:34:13]E (63623) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[22:34:13]E (63623) task_wdt:  - loopTask (CPU 1)
[22:34:13]E (63623) task_wdt: Tasks currently running:
[22:34:13]E (63623) task_wdt: CPU 0: IDLE0
[22:34:13]E (63623) task_wdt: CPU 1: IDLE1
[22:34:13]E (63623) task_wdt: Aborting.
[22:34:13]
[22:34:13]abort() was called at PC 0x4203cd50 on core 0
[22:34:13]
[22:34:13]
[22:34:13]Backtrace: 0x40377cae:0x3fc958c0 0x4037cc79:0x3fc958e0 0x40382a35:0x3fc95900 0x4203cd50:0x3fc95980 0x40379361:0x3fc959a0 0x420cf3a3:0x3fcf4840 0x4203cadd:0x3fcf4860 0x4037e220:0x3fcf4880
[22:34:13]
[22:34:13]
[22:34:13]
[22:34:13]
[22:34:13]ELF file SHA256: d4a54dfa1d4ace1a
[22:34:13]

GabrielSimoes1983, 2025-04-29:
Good news: I switched 2 pins: 3v3 power and the TX pin and now it works (still don´t know what was wrong, probably some issue with the TX 43 pin).
Erros still here...

RowanTaubitz, 2025-05-04:
Turn off "Auto Updates" and it will stop crashing. There's something wrong with checking github for an update and it causes it to lock up.

I was getting the same error, always after the `Check github for updates...` message:

Check github for updates...
E (70177) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (70177) task_wdt:  - loopTask (CPU 1)
E (70177) task_wdt: Tasks currently running:
E (70177) task_wdt: CPU 0: IDLE0
E (70177) task_wdt: CPU 1: loopTask
E (70177) task_wdt: Aborting.

abort() was called at PC 0x40113a18 on core 0

Backtrace: 0x40083d25:0x3ffbed6c |<-CORRUPTED

GabrielSimoes1983, 2025-05-06:
Thanks! I will try that