rstrouse/ESPSomfy-RTS #571: ESP Somfy RTS unresponsive to commands (UI/HomeAssistant)

chenganir, 2025-05-12:
### Hardware

ESP32

### Firmware version

v2.4.6

### Application version

v2.4.6

### What happened? What did you expect to happen?

I managed to pair 4 Somfy RTS roller shades, and was able to link remotes, and scan for frequency and set it (433.42). However, there seems to be some kind of delay or unresponsiveness when pressing UI elements (up/down) or even from home assistant (up/stop/down). Seems like I need to hit the button repeatedly multiple times to see some action. 

I do not have any of these issues with the original remotes, a wall plate remote, and the Somfy MyLink.

Would love to provide more info and logs, let me know how I can collect them. 

Thanks.

### How to reproduce it (step by step)

```bash
1. Go to...
2. Click on...
...
```

### Logs

```bash

```

awrede, 2025-06-08:
I also see this problem. I have to send multiple (4 to 8) up or down commands before the cover moves. Subsequent commands are executed every time, until the pause between two commands is longer than approximately 8 to 10 seconds. After that pause it will take 4 to 8 commands to get a response.

chenganir, 2025-06-09:
Happy to provide logs if needed, I can reproduce this always using either the webUI or homeasssitant. Just provide the instructions to collect whatever you need

bradhs, 2025-06-09:
Unfortunately I wont be much help as I am less than 24 hours into this repo.

Nickduino, 2025-06-21:
Have you tried increasing the Repeat?

[image]
(20 is probably way too much, it's just an example)

chenganir, 2025-06-21:
Thanks @Nickduino , setting it to 20 did resolve the issue! I'll check this again a few times and then try to tweak it to a lower value.