rstrouse/ESPSomfy-RTS #573: ESP32-C6 Zero Mini

matthieuneier, 2025-06-09:
### Hardware

ESP32

### Firmware version

2.4.6

### Application version

2.4.6

### What happened? What did you expect to happen?

Hello I try to install ESPSomfy-RTS on a ESP32-C6 Zero Mini with web.esphome.io
I was able to flash the SomfyController.onboard.esp32.bin but WIFI AP does not appear
I also try the esp32c3 and esp32s3 version without success
What the correct version for this chip ? does the C6 Zero Mini supported with ESPSomfy-RTS ?
Many thanks

### How to reproduce it (step by step)

```bash
1. Go to...
2. Click on...
...
```

### Logs

```bash

```

matthieuneier, 2025-08-25:
Hello, I update the hardware I used in the issue.
After this misadventure, I bought a Waveshare ESP32-C3 Mini 
Which is the previous generation of this chip, and it works perfectly

384KB ROM
400KB HP SRAM
??KB LP SRAM
4MB Flash

sitexw, 2025-08-25:
Thanks for your reply and for the detailed info you shared about your hardware, that’s really helpful!

On my side, the ESP32-S3 Zero starts up and works, but it ends up crashing whenever I use it a bit too much. For example, if I raise/lower the blinds too often, it seems to saturate something, then it stops responding and eventually reboots on its own after a while (or if I unplug/plug it back in, it immediately works again). So that’s what makes it unusable for me right now.

For reference, I’ve ordered another ESP32-S3, this time with more memory (16 MB flash and PSRAM), because I thought the issue might be related to insufficient RAM. But as you mentioned, your C3 works fine even with much less memory (not even 512 KB), which I find quite surprising.

Anyway, I’ll test with my new board as soon as it arrives and will report back whether it fixes the problem or not. Worst case, I’ll also give the C3 a try, since it seems to work well for you.