rstrouse/ESPSomfy-RTS #577: WLAN fluctuations

christianhcm, 2025-06-26:
### Hardware

ESP32

### Firmware version

v2.4.6

### Application version

v2.4.6

### What happened? What did you expect to happen?

I am observing Wi-Fi fluctuations even to the point of of loss of connection.  It started on 23.06.2025 .This also occours with a second identical ESP 32.Swapping cable and power source and disabling roaming hasn't changed anything. This renders all automations inoperable, as  the WI-Fi connection drops during execution.

### How to reproduce it (step by step)

```bash
1. Go to...
2. Click on...
...
```

### Logs

```bash

```

sitexw, 2025-08-20:
Hi,

Thanks for sharing the details of your problem. I’ve been following similar reports and in some cases the issue might not be purely Wi-Fi related, but rather connected to memory limitations on certain ESP32 boards (running out of RAM can sometimes trigger resets or unstable behavior).

To help compare experiences, could you share the exact specifications of the boards you tested (model, amount of RAM/PSRAM, flash size, etc.)?
That would make it easier to see whether these Wi-Fi drops are linked to hardware limitations or if something else is going on.

Thanks!

abaghbani, 2025-10-21:
Hello, I have this esp32-s3:
https://www.wemos.cc/en/latest/s3/s3_pro.html
based ESP32-S3-WROOM-1
16MB Flash (Quad SPI)
8MB PSRAM (Octal SPI)

and just try rev 2.4.7 (latest) and got this error:

load:0x3c0e0020,len:0x33e0c
load:0x3fc945c0,len:0x577c
load:0x40374000,len:0x6a60
load:0x42000020,len:0xd5574
load:0x4037aa60,len:0x9b5c
SHA-256 comparison failed:
Calculated: 968eb62d9cd86529a9b1c33173605786d532790e5af79ef654c7a27941920b95
Expected: 4c52da676c17c56754d6fd02776e5a04d0ea4b9f794007e24c84b229ff3d636a
Attempting to boot anyway...
entry 0x40377440
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403743c0
SPIWP:0xee
mode:DIO, clock div:1
load:0x3c0e0020,len:0x33e0c
load:0x3fc945c0,len:0x577c
load:0x40374000,len:0x6a60
load:0x42000020,len:0xd5574

I also tested with ver2.3.3, similar issue:

load:0x4037af84,len:0xa450
load:0x4037af84,len:0xa450
SHA-256 comparison failed:
Calculated: 63deb6dcb0087bd1a33bb88bc1dbc4567779655daca36ecc23155882c23cbb4b
Expected: 0328425b7288d1f02b4a4efa865d964bb54dc11db855be422c0dfb15efb86cea
Attempting to boot anyway...
entry 0x40377470
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40379adb
SPIWP:0xee
mode:DIO, clock div:1
load:0x3c0e0020,len:0x33988
load:0x3fc953e0,len:0x56dc
load:0x40374000,len:0x6f84
load:0x42000020,len:0xd2028

not sure is it related to flash/ram size issue or not

Dschuli, 2026-09-18:
Same here. Started (or I noticed it) about 3 month ago. ESPSomfy RTS devices in HA drop out for seconds to minutes. ESP32 (standard devkit)  running V2.4.6 becomes unresponsive to pings as well - which strongly suggests a WLAN instability.
Can share device availabilty and ping logs - but they just show the issue. 
Normally,  after reboot there is a stable period before the issue returns. 
Tried power cables & supplies; moved the device to a SSID with only this device  - no long term improvement.  
Opening the Web UI sometimes triggers the problem.  
Free memory always around 160k. 

As a stop-gap I installed an external watchdog (ESP with relay) that reboots the ESPSomfy via a power cut when a device becomes unvailable for a longer period (3mins) - but would rather solve the underlying issue.

Any help/idea welcome.

Dschuli, 2026-09-22:
Checked the router log (Gl-iNet MT6000) and found many entries like the below

13:20:48 DHCPDISCOVER f4:65:0b:49:4b:0c
13:20:48 DHCPOFFER   10.0.1.251
13:20:48 DHCPREQUEST 10.0.1.251
13:20:48 DHCPACK     10.0.1.251 ESPSomfy-RTS

13:36:19 DHCPDISCOVER f4:65:0b:49:4b:0c
13:36:19 DHCPOFFER   10.0.1.251
13:36:19 DHCPREQUEST 10.0.1.251
13:36:19 DHCPACK     10.0.1.251 ESPSomfy-RTS

14:44:26 DHCPDISCOVER f4:65:0b:49:4b:0c
14:44:26 DHCPOFFER   10.0.1.251
14:44:26 DHCPREQUEST 10.0.1.251
14:44:26 DHCPACK     10.0.1.251 ESPSomfy-RTS

14:55:39 DHCPDISCOVER f4:65:0b:49:4b:0c
14:55:39 DHCPOFFER   10.0.1.251
14:55:39 DHCPREQUEST 10.0.1.251
14:55:39 DHCPACK     10.0.1.251 ESPSomfy-RTS

15:07:49 DHCPDISCOVER f4:65:0b:49:5f:ac
15:07:49 DHCPOFFER   10.0.1.251
15:07:49 DHCPREQUEST 10.0.1.251
15:07:49 DHCPACK     10.0.1.251 ESPSomfy-RTS

So the router sees the ESP repeatedly starting DHCP from scratch, but it does not log a corresponding deauth, disassoc, or MacTableDeleteEntry for that MAC.
That points more toward the ESP itself restarting/reinitializing Wi-Fi or its DHCP client than the router explicitly kicking it off.