rstrouse/ESPSomfy-RTS #578: "Press and hold" to pair does not work for some shades that need it

syko, 2025-06-26:
### Hardware

ESP32

### Firmware version

v2.4.7

### Application version

v2.4.7

### What happened? What did you expect to happen?

Hey, found a small issue but one that breaks pairing for certain shades.

The pairing screen reads `NOTE: On some motors you may need to press and hold the Prog button` but this does not work (at least on some shade models).

The reason is that the UI does not send any `repeat` parameter to the backend and if the shade does not have repeat frames configured then long-pressing the pair button does not produce real repeat frames but multiple normal frames instead.

The shades I have are operating on 80-bit RTS so the motor expects the repeat frames to have 6 hw sync pulses instead of the inital 12 but long pressing produces only frames with 12 pulses and it looks like this is enough to break everything.

**Fortunately there's a workaround**: before pairing, configure some "Repeat Commands" for the shade and just tap the Pair button. Something between 2 and 10 worked for me.

### How to reproduce it (step by step)

```bash
1. Acquire some shades that need multi-frame pairing and which care about the correct sequence of hw sync pulses
2. Go to shade pairing screen
3. Long-press the Pair button
4. Observe utter lack of pairment
```

### Logs

```bash

```

Peethiplatsch, 2025-07-15:
I think we have more or less the same issue.

https://github.com/rstrouse/ESPSomfy-RTS/issues/583

otaviobonder-deel, 2025-09-15:
That worked for me too, thanks!