rstrouse/ESPSomfy-RTS #579: Web page not accessible "Error opening file"

synop6, 2025-06-30:
### Hardware

ESP32

### Firmware version

v2.4.6

### Application version

v2.4.6

### What happened? What did you expect to happen?

Hello All, 

I tried to install the last stable firmware on a ESP32, flashing seems to passing well but once it's ended and i try to reach the interface via 192.168.4.1 i have an error message who tells "Error opening file"
I tried on different ESP32, same result. 
I used ESP Home flasher, type of ESP on logs provided. 

I also tried the others firmware but seems to be not compatible with my EPS model 

Is someone know what is this error ? 
In the past i succeeded to install it on other ESP that burned due to a bad power supply ... now i'm not able to install it properly 

Thanks to all for your help 

### How to reproduce it (step by step)

```bash
1 download firmware SomfyController.ino.esp32.bin
2 go to esp home flasher select serial port, firmware and flash
```

### Logs

```bash
Using '/dev/cu.usbserial-10' as serial port.
Connecting.....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting..........
Detecting chip type... ESP32
Connecting...........

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32-D0WD-V3 (revision 3)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: 14:33:5C:30:85:E8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 4MB
 - Flash Mode: dio
 - Flash Frequency: 80MHz
Erasing flash (this may take a while)...
Chip erase completed successfully in 1.5s
Flash will be erased from 0x00001000 to 0x00004fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x0014efff...
Compressed 15856 bytes to 10276...
Writing at 0x00001000... (100 %)Wrote 15856 bytes (10276 compressed) at 0x00001000 in 0.3 seconds (effective 374.1 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 144...
Writing at 0x00008000... (100 %)Wrote 3072 bytes (144 compressed) at 0x00008000 in 0.0 seconds (effective 614.5 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 1639.1 kbit/s)...
Hash of data verified.
Compressed 1305536 bytes to 811051...
Writing at 0x0014bcfb... (100 %)Wrote 1305536 bytes (811051 compressed) at 0x00010000 in 21.1 seconds (effective 495.8 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Hard resetting via RTS pin...
Done! Flashing is complete!

Showing logs:
[12:06:09]E (445) esp_core_dump_flash:`ɕdump partition found!
[12:06:09]E (445) esp_core_dump_flash: No core dump partition found!
[12:06:09]E (299
[12:06:09]Startup/Boot....
[12:06:09]Mounting File System...
[12:06:09]E (309) esp_littlefs: ./components/esp_littlefs/src/littlefs/lfs.c:1367:error: Corrupted dir pair at {0x0, 0x1}
[12:06:09]
[12:06:09]E (310) esp_littlefs: mount failed,  (-84)
[12:06:09]E (314) esp_littlefs: Failed to initialize LittleFS
[12:06:09]Error mounting file system
[12:06:09]Chip Model ESP32-
[12:06:09]Preference IP Free Entries: 621
[12:06:14]SECURITY   Type:0 Username:[] Password:[] Pin:[] Permissions:0
[12:06:14]Connection Type: 1
[12:06:14]NTP Settings 
[12:06:14]pool.ntp.org TZ:
[12:06:14]WIFI Settings
[12:06:14] SSID: [] PassPhrase: []
[12:06:14]
[12:06:14]Launching web server...
[12:06:14]Creating Web MicroServices...
[12:06:15]WiFi Mode: 0
[12:06:15](evt) WiFi interface ready
[12:06:15]WiFi station mode started
[12:06:15]Socket Server Started...
[12:06:15]App Version:0.0.0
[12:06:15]Starting clean
[12:06:15]Initializing RX Queue
[12:06:15]
[12:06:15]Turning the HotSpot On
[12:06:15](evt) WiFi SoftAP Started IP:192.168.4.1
[12:06:15](evt) WiFi SoftAP Stopped
[12:06:15](evt) WiFi SoftAP Started IP:192.168.4.1
[12:06:15]Timing Net: 203ms
[12:07:04]Loading file /index.html
[12:07:04]Error opening/index.html
```

sitexw, 2025-08-20:
Hi,

Thanks for sharing the details of your issue. I’ve been following several similar cases, and what I’ve noticed is that sometimes the behavior depends a lot on the exact hardware specs of the ESP32 being used (available RAM/PSRAM, flash size, etc.).

Would you mind sharing the exact specifications of the boards you tested (model name + how much RAM/PSRAM + flash they each have)?
That would help build a clearer picture of which boards work reliably and which ones tend to run into problems.

Thanks a lot!

chathamws, 2025-12-25:
Same issue for me after compiling/uploading from source code to an ESP32-WROOM-32E. After flashing, connecting via Wi-Fi, and opening a browser to 192.168.4.1, the browser displays "Error opening file". I used the 1.8.x version of the Arduino IDE, as mentioned in the doc, and all the required libraries were present. I chose the manual route because the basic flashing steps (online or with flashtool) using the bin file do not work on this device. After flashing, it's stuck in a boot loop / green light constantly flickering.

simonthechipmunk, 2026-09-28:
I just had the same issue and the solution is exactly what I suspected when I read the error message:

On new devices you usually need to flash the bootloader, partition table, firmware and in this case the littleFS file system as well. All of these need to be written to their respective addresses in flash.
If you download the ino.bin file for your ESP directly from the release page it will only contain the firmware, nothing else.

Download the onboarding zip file instead and extract the bin file from there. It's significantly larger (~4Mb instead of 1Mb) and contains all the individual parts I mentioned above in a single bin file that you can flash in one go.

Rooky mistake and basically an RTFM issue. I think this can be closed as it's not a bug.