rstrouse/ESPSomfy-RTS #581: assembled kit

marsonpl, 2025-07-06:
Are there any pre build kit with this project? I have somfy blinds that uses RTS, but unfortunately I am disabled  so even easiest soldering is out of of my scope, I happily pay for assembled kit with this project ready to use

watgaan, 2025-07-06:
Hi,

My current RTSSomfy I used a ESP32 with pins already on the board. No soldering needed. the CM11 module also has pins. Only thing I did was connect it all up with DuPont Jumper wire Female-Female 10cm 10 wires (https://www.tinytronics.nl/en/cables-and-connectors/cables-and-adapters/prototyping-wires/dupont-compatible-and-jumper/dupont-jumper-wire-female-female-10cm-10-wires).
Loaded the software (to me a not straight forward process) and connected to the device with a phone to setup IP and WiFi network.

Now I am building a version that connects to ethernet with PoE (Lilygo), In soldering stage then I need to work out how to load software (yes I have the UART module or whatever it is called). I hope to get this done before August...

bradhs, 2025-07-26:
I just built one of these on a breadboard.  Works perfectly.  Now just need someone to solder this onto a circuit board or build one for me. :)

marsonpl, 2025-07-29:
My colleague assembled it for me and printed 3D case then sent all ready to use, the device itself works nicely and it is easy to setup.

closed 2025-07-29 by marsonpl