rstrouse/ESPSomfy-RTS #587: Unable to have esp32-c6 or s3 to boot.

dbn-b4e, 2025-07-26:
Hello,

I flahed all binaries on esp32 s3 and it does not boot is in boot loop. Esp board is not the problem.

Anybody faced this or have any idea?

Thx

watgaan, 2025-07-26:
Yes I have had this also, as I am no expert at all on ESP32 I managed to fix this by loading the image to 0x0 position. Thanks to Chat in helping out I used Python ESPtool to load it. First erase flash:
esptool.exe --chip esp32 --port COM5 erase_flash, off course use your own COM port
esptool.exe --chip esp32 --port COM5 --baud 460800 write_flash -z 0x0 SomfyController.onboard.esp32.bin
That did the trick for me loading it to 0x1000 had mine going in boot loop.
When rebooted you can connect to the ESPSomfy WiFi access point (default IP 192.168.4.1) to configure WIfi etc. 
Hope this helps!

watgaan, 2025-07-29:
And did you get firmware/binaries loaded?

neilbrookins, 2025-08-05:
On my s3 it does not boot when the power to cc1101 is connected. But with radio disconnected it boots ok.  I have not been able to resolve this issue.  So I just bought a regular older model esp32 and it's working great.  I read somewhere that some esp32-s3 devices may use lower voltage like 1.8v so perhaps that is related? I don't know.  I'm ok with my work around.

leedizzal, 2025-08-28:
Hopefully it was helpful, let us know if you got it going... 

Lee

dbn-b4e, 2025-08-28:
Had no time today will try tmr and surely let you know

celegroz, 2026-02-01:
I'm having the same problem with Xiao esp32c3. The strange thing is that if I keep powering off/on the esp, eventually I can get it to boot up. But, it is very unreliable/random on when it works. Any ideas?