rstrouse/ESPSomfy-RTS #592: Pairing a Shade/Motor not working when I click on Prog button (UI Interface)

AntMarq, 2025-08-17:
### Hardware

ESP32

### Firmware version

v2.4.6

### Application version

v2.4.6

### What happened? What did you expect to happen?

Hi,

I have an ESP32 and a C1101 sensor. I installed the software ESPSomfy-RTS via HACS.
After following the steps, I was able to log the inputs from Somfy remotes.

Next, I created a shade with a random value in the "type" field, 56 bits.

[image]

[image]

When I try to pair, pressing the physical button on the remote for 2 seconds does make the shutter jog up and down. However, clicking the "Prog" button in the UI afterwards does nothing.

I've tried multiple times without success.

I read that one possible solution is to do a factory reset on my Somfy shutter. Is there any risk in doing that? Could I be missing something?

Thanks for your help.

PS : I haven’t been able to find or retrieve the logs from my HA/Raspberry Pi 4. If you know where I can find them, I’ll gladly add them to the ticket.

### How to reproduce it (step by step)

```bash
Unable to reproduce without the hardware
```

### Logs

```bash

```

leedizzal, 2025-08-28:
Are you sure you transceiver is wired correctly and working? Did you verify all the GPO pins configured match their respective connections to the transceiver. Perhaps you have the tx pin miss configured... Have you tuned the radio to get the best signal from the RTS remotes?

leedizzal, 2025-09-03:
So as I was saying you do not have the transceiver properly connected to your ESP32 board. You have inverted the TX and RX pins based on the information you just posted. So you either need to invert the connections to pin 3 and 8 on the C1101 board or simply invert the GPIO pins in the software configuration so make TX GPIO-12 and RX GPIO-13 and that will fix your problem. 

You can receive signals with the pins inverted but are not going to be able to transmit signals. 

Let me know if that solves your issue... 

Cheers!

Lee

AntMarq, 2025-09-04:
I have switch the GPIO pins in the software and it's work ! Thank for your help 🙏

jackshaw20, 2025-09-13:
Hi, was the problem resolved as I have the same problem.   

If I revert my pins the blind will operate but it reboots the ESP32 each time. 

If I wire it correctly I can only receive signal from the remote and nothing hets transmitted 

Could it be a fault C1101?

jackshaw20, 2025-09-14:
here is the log;

[08:37:24]Sending Shade Command
[08:37:24]Received:{"shadeId":1,"command":"down"}
[08:37:24]CMD:Down ADDR:51310 RCODE:113 REPEAT:0
[08:37:24]Enabled receive on Pin #12 Timing: 1
[08:37:24]Timing WebServer: 167ms
[08:37:32]E (95516) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[08:37:32]E (95516) task_wdt:  - loopTask (CPU 1)
[08:37:32]E (95516) task_wdt: Tasks currently running:
[08:37:32]E (95516) task_wdt: CPU 0: IDLE0
[08:37:32]E (95516) task_wdt: CPU 1: IDLE1
[08:37:32]E (95516) task_wdt: Aborting.
[08:37:32]
[08:37:32]abort() was called at PC 0x40113954 on core 0
[08:37:32]
[08:37:32]
[08:37:32]Backtrace: 0x40083d25:0x3ffbed6c |<-CORRUPTED
[08:37:32]
[08:37:32]
[08:37:32]
[08:37:32]
[08:37:32]ELF file SHA256: f81fd5f3694043bf
[08:37:32]
[08:37:32]Rebooting...

RobinEd, 2025-10-02:
Same happened to me with an E07-M1101D and ESP32.

I initially used the pin order in picture https://user-images.githubusercontent.com/47839015/227078539-9b7a5f0d-21a4-45c5-b96b-cabfaceafcbb.png for the pinout (under https://github.com/rstrouse/ESPSomfy-RTS/wiki/Simple-ESPSomfy-RTS-device - ESP-WROOM-32/ESP32S Models). 

The RX/TX pins (pin 8 and 3) in the pin list under the transceiver pins are mixed up in that picture. Here is a fixed version.

[image]

.