rstrouse/ESPSomfy-RTS #598: LILYGO T-Embed ESP32-S3 CC1101 not detecting signal

brycetg, 2025-09-06:
### Hardware

ESP32-S3

### Firmware version

v2.4.7

### Application version

v2.4.7

### What happened? What did you expect to happen?

When scanning frequency using the web interface, no signal detected when pressing buttons on remote. (RSSI remains blank). Have tested hardware using BRUCE and was able to detect a signal on 433.420MHz. Have confirmed blind motor is Somfy.

### How to reproduce it (step by step)

```bash
Configured pins as per https://github.com/Xinyuan-LilyGO/T-Embed-CC1101
```

### Logs

```bash
[12:09:25]Applying radio settings Setting Data Pins RX:38 TX:3
[12:09:25]Setting SPI Pins SCK:11 MISO:10 MOSI:9 CSN:12
[12:09:25]Radio Pins Configured!
[12:09:25]Successfully set up the radio
[12:09:25]Enabled receive on Pin #38 Timing: 1
```

brycetg, 2025-09-06:
[image]

brycetg, 2025-09-16:
> Oh that is no good I am building one on that platform to replace my other one powered by USB. When building that one I had same issue. Solution was to correct wire the CC1101 to the board. I mixed up 2 wires after adjusting so it matched, all was well.

@watgaan do you mean the T-Embed arrived incorrectly wired? Or were you building from scratch? The one I bought already had the CC1101 wired in - wondering if I need to reassemble it

r-sidshaytay, 2026-11-06:
We have this working on a T-Embed CC1101 Plus and have submitted [a board-support PR to cjkas’s fork](https://github.com/cjkas/ESPSomfy-RTS/pull/133). It handles the RF switches mentioned above, peripheral power and shared-SPI startup. The PR is still open, not merged.

The standard T-Embed CC1101 should work with the same profile—the documented wiring and memory match—but we’ve only tested the Plus ourselves. If anyone here has the standard model, please try it and report your board revision and results in the PR. Take a fresh backup and follow its build/migration instructions, since this is based on cjkas’s 4.x firmware.